gnuradio/gnuradio #1242: Migrate to MediaWiki

mbr0wn, 2017-03-16:
We're already pretty far: http://wiki.gnuradio.org

And we want to get off of Redmine ASAP.

This is a list of things we want to fix on the new wiki. Note they won't block us from switching off Redmine.

- [ ] Enable SSL
- [ ] Enable github OAuth

wgebers, 2017-03-18:
I foresee a need for the following: 
- A place to store files, for example, pdf's, documents, grc files.
-- Specifically, .doc / .pdf for the academic papers page.  
-- tar.gz for specific software configurations that are not quite tutorials. 
-- It may be simplest to have a sub-folder / repository on github that handles 'resources' rather than adding additional file types to the wiki.

- A link on the left hand side menu bar to the 'All Pages' page.

mbr0wn, 2017-03-19:
Added links on the left. Do you really think we want to link to "All Pages" though? There's a) the search and b) you get this via special pages -> all pages. Might clutter the sidebar.

Do you want to give the style guide a shot?

wgebers, 2017-03-19:
Will work on the style guide. 

My thought with the 'All Pages' link was to make it really easy for someone with no MediaWiki experience and new to GnuRadio to see what is available in terms of resources.  It probably is a duplication.

mbr0wn, 2017-03-20:
OK, PDF upload is now enabled. I'm not so sure about .tar.gz, but it can be added fairly easily.

Thanks for taking a stab at a style guide! Also, your suggestions are highly appreciated. I wasn't trying to shoot down the "All Pages" suggestion, and if there's demand it's easy to add to the sidebar now.

mbr0wn, 2017-03-20:
...we're off of Redmine at this point.

bastibl, 2017-03-20:
Maybe the "Development" tab on gnuradio.org should be renamed to Wiki? If someone is new to GNU Radio and looking for tutorials and stuff like that, he might not expect it under development.

jmcorgan, 2017-03-21:
I suggest we change the title of the main page to something besides "Main Page", like "Welcome to GNU Radio".  This would let us remove the extra header in the actual content.

mbr0wn, 2017-03-21:
@jmcorgan I've removed the extra header. Need to double-check about renaming the main page.

@bastibl Yeah I thought the same. Need to figure out how to change stuff on the website.

jmcorgan, 2017-03-21:
The main page does look cleaner now, thanks.

devnulling, 2020-01-19:
Closing this as the migration is complete and a lot of the pages have been added to various categories.

closed 2020-01-19 by devnulling