gnuradio/gnuradio #1244: gr_modtool: Failed import in Python QA

mhostetter, 2017-03-20:
I have an OOT module called `gr-misc`. This is the auto-generated Python QA if I add a block called `blah`.
```
from gnuradio import gr, gr_unittest
from gnuradio import blocks
import misc_swig as misc

class qa_blah (gr_unittest.TestCase):

    def setUp (self):
        self.tb = gr.top_block ()

    def tearDown (self):
        self.tb = None

    def test_001_t (self):
        # set up fg
        self.tb.run ()
        # check data

if __name__ == '__main__':
    gr_unittest.run(qa_blah, "qa_blah.xml")
```

Python complains saying:
```
matt@t430:~/repos/gr-misc/python$ ./qa_blah.py 
Traceback (most recent call last):
  File "./qa_blah.py", line 24, in <module>
    import misc_swig as misc
ImportError: No module named misc_swig
```

I'm not sure why it's needed, but just wanted to point it out.

mhostetter, 2017-03-20:
So, I believe the `import misc_swig as misc`, again where `gr-misc` is the GNU Radio module, is to import the module under test.  On my machine, at least, `import misc` works but the first does not.

jmcorgan, 2017-03-20:
Those scripts aren't meant to be run directly; rather, they are executed when running 'make test' in the build directory.  In that case a very special environment is setup such that the uninstalled code in the build directory is used instead of anything the might have already been installed on the path.  That is why the import fails when running it in a normal environement.

mhostetter, 2017-03-20:
Ah, ok! No issue then. Thanks for replying and clarifying for me.

closed 2017-03-20 by mhostetter