gnuradio/gnuradio #1249: grc: Add units column to block properties dialog

mhostetter, 2017-03-23:
This a feature request. Just a thought I had and I wanted to share it.

I think it would be nice to add a Units columns to the block Properties dialog box. I was thinking as the 3rd column, to the right of the parameter value entry column. Often times when making a parameter I feel the need to add the units in parentheses to the `<name>` tag, but the name gets too long and it becomes cluttered. It would be nice to have a standard interface.

We could add an optional `<unit>` tag inside `<param>`. It could default to an empty string if not included. Some units that I think would be helpful are `Hz`, `s`, `samples`, `channels`, `%`, `dB`, `linear`. I think this would make blocks easier to understand/use for a general user -- someone who hasn't looked at the source code to figure out how the block works or what units its expecting.

Some examples of confusions I see in blocks are:

* Percentages -- is the block expecting units of % or as linear, e.g. if I want 10.0 % do I enter `10.0` or `0.10`
* Periodicity or rates -- is the block expecting units of Hz or s
* Thresholds -- is the block expecting linear units or dBs

I think by making it an optional tag, it would be backwards compatible with old blocks. And we (I) can slowly add the appropriate units to each block.

Thoughts?

jmcorgan, 2017-03-23:
I think this is a great idea.  The GRC team is a little stretched at the moment working on 3.8, but if you'd be willing to help at their direction on this, I think this could be done in our current code base.

jmcorgan, 2017-03-23:
@skoslowski

mhostetter, 2017-03-23:
I'm always willing. I really know nothing about the code architecture for GRC and I don't have tons of QT experience, but if @skoslowski can point me in the right direction, I'm willing to take a crack at it.

skoslowski, 2017-03-23:
this could potentially take up too much space, e.g. if sth. "channels" is used. 
- I'd rather have that information shown as part of #1250, maybe using special formatting to highlight it
- or show param names and that unit in a separate row, on the left and right, respectively. would have to try that out to see how it looks

awalls-cx18, 2017-03-30:
Some units can be difficult to state tersely. "normalized radians/sample" (PLL blocks) and "nominal samples/second" (symbol timing recovery blocks) come to mind.
But overall, I like the idea.

jmcorgan, 2017-04-02:
@skoslowski I'll assume you'll take lead on figuring out how to get this implemented and into a PR.

marcusmueller, 2023-10-24:
I'd still love to see this; #6888 removed the "implicit documentation" through colors (which had downsides on accessibility, and on dark themes) and made it explicit in an extra column. On modern screens, horizontal space is really not an issue.

marcusmueller, 2023-10-24:
I'm leaving this very well-aged issue open, as it has relevance for GR4.0 GUI