gnuradio/gnuradio #1250: grc: Add hover-over documentation for block parameters

mhostetter, 2017-03-23:
This is a feature request.

I think it would be nice to add another documentation feature to blocks in GRC. When I'm using blocks in GNU Radio, often times I don't know the exact intent of a particular parameter. Take the Clock Recovery MM block, for instance, a block I'm currently trying to master.

[image]

As a general user, I don't really know what `Mu` or `Gain Omega` are or how I should/shouldn't manipulate them. It would be really nice to get some detail about exactly what that parameter means and how to modify it...in GRC and in the block properties window.

My thought on how to implement this is to add an optional `<doc>` tag under `<param>`, similar to the `<doc>` tag for the whole block itself. It would default to an empty string if not defined, but would allow a developer to add quite a bit of detail explaining that parameter if he desired. GRC would then query that tag in an `on_hover()` callback of some sort. (I'm not sure how hard the hover callback is)

Some parameter meanings are obvious, but there are several out there in GNU Radio that are quite mysterious. I think this would allow us to add some detail to help bridge the gap between developer and user in a way that's easily accessible to the user.

I know the documentation is pretty good on the doxygen page, but general, lay users aren't navigating there in my opinion. Having more documentation right inside the block properties tab is very useful in my opinion.

Thoughts?

jmcorgan, 2017-03-23:
My comments in #1249 apply here as well--except here we are supposed to have doxygen param comments in the block header files for this.  These could get propagated to the block/param documentation in GRC like the rest do.  The <doc> section in XML is really only for unusual blocks that don't have this mechanism, and is mostly unused.

mhostetter, 2017-03-23:
Ok, if there's a mechanism to access the doxygen `\param` documentation from GRC, that would work.

skoslowski, 2017-03-23:
Per param documentation would be quite nice. I had thought this before, when the docstring loader was rewritten. here are my notes on it:

- Currently, GRC attempts to guess the module and class name of each (swig-wrapped) block. It then loads that and reads the class docstring which gets placed there by swig. that happens async, triggered by a reload of the block tree.

- It would be nicer to do this only once, maybe even at build time. By using the doxygen (XML) output, matching classes to XML files and extending them.

- trouble is getting the information into GRC, display could be simply a tooltip

marcusmueller, 2023-10-24:
Since this issue was opened, the block docs have essentially been crowd-sourced through the wiki, as the doxygen docs have an API/developer, not a user focus. That's been a great success! However, it means that such information is hard and manual to come by.

I'm not sure it's a good idea to close this issue – we should have docs being available from the properties dialog – but with the shape these docs have now taken, the parameter documentation often is less useful without context. I could very well imagine a splitscreen kind of approach could work well, where interacting (context menu? clicking on the name) with a parameter row would scroll to /highlight the lines documenting the parameter as imported as whole wiki page.

Considering the age of the issue and the amount of work to get these structured info out of the current documentation back into GRC reliably are right now prohibitive, I'm closing this with a heavy heart.

closed 2023-10-24 by marcusmueller