gnuradio/gnuradio #1261: qa: flaky/broken unittests

noc0lour, 2017-04-04:
A couple of unit tests are  flaky and/or broken. In favor of opening a separate issue for each one I'd collect them here in a check list. 

Everyone is welcome to investigate the root cause for a flaky/broken QA test, just drop a comment down below which one you are looking into. A lot of times it is just a race condition when starting/stopping a flow graph, so fixing these should be fairly straight forward.

- [ ] qa_header_payload_demux (test_001 contains nodata when it should)
- [ ] qa_udp_source_sink (test_002 contains no data when it should)
- [ ] qa_hier_block2 (segfault in test_34b)

jmcorgan, 2017-04-04:
The qa_pmt segfault only happens on the python3 branch.

jmcorgan, 2017-04-04:
Also, just FYI, most of the above issues only happen with parallel ctest invocations; that is, the underlying issues exist but aren't often triggered when doing plain 'make test'.  To see the above, run 'ctest -j <ncores>' in the build directory.

noc0lour, 2018-02-04:
So, people have a hard time packaging GNU Radio with flaky and broken tests. Ideally we fix the underlying issues (or the flakiness of the tests if it's a test issue). Other ideas? I thought about removing the "real" flaky tests und replace them with non-flaky ones.

marcusmueller, 2018-02-11:
We can't make tests that actually demonstrate a bug pass just so that people can package stuff. There *are* bugs in GNU Radio, *even in packaged versions*.

What we *can* do is make the tests we know fail non-fatal. But I really don't know whether that helps anyone – I'd actually like to ask Mait about what's to do from a debian perspective about failing `make test`.

noc0lour, 2018-02-11:
There is a three types of test: 
 - 1. fails consistently if there is a bug in the implementation -> good test
 - 2. fails unpredictably -> worthless
 - 3. Test has a bug and fails unpredictably -> bug in the test 

Separating the last two and rewriting the 2. from scratch and fixing the 3. would be the way to go. Thus I collect these tests here.

marcusmueller, 2018-03-08:
As per discussion today: The intended way to move forward is to remove these tests, and open a bug report for every removed test.

I wonder if we can deduct from this one bug report a series of similar pull request? I feel that for every test we remove from the tree, we should have a separate issue "digital: removed flaky qa_milk. digital::moloko::plus now untested" that we can work on.

noc0lour, 2018-03-15:
Running 1000 tests with `-j20` yields this list of failing tests:
```
╰─ grep -i \*failed /tmp/gr-test-stdout | awk -F' ' '{print $4}' | sort -u
qa_constellation_receiver
qa_constellation_soft_decoder_cf
qa_fastnoise
qa_filterbank
qa_flowgraph
qa_header_payload_demux
qa_hier_block2
qa_moving_average
qa_ofdm_sync_sc_cfb
qa_packet_format
qa_packet_headerparser_b
qa_udp_source_sink
qa_zeromq_pub
```

marcusmueller, 2018-03-22:
@noc0lour so, how to proceed with this? Got time to sit together and do the above, ie. remove flaky test, open issue for lacking test coverage?

noc0lour, 2018-03-26:
I'll take a look today what we can do to those unittests.

marcusmueller, 2018-03-26:
@noc0lour is the qa_zeromq_pub lockup solved by adding delays , like #1702?

noc0lour, 2018-03-26:
Hopefully, though the patch in #1702 is for `_sub`

marcusmueller, 2018-03-26:
Yeah was referring to the same approach

marcusmueller, 2018-03-28:
@noc0lour I have need to release 3.7.12 Friday.

How can we proceed with this? Since unit test exclusion doesn't change ABI (unless we do something stupid!), we could do this by 3.7.12.1 ?

noc0lour, 2018-03-28:
yup, we can postpone this, or someone else wants to figure out which tests to remove and why

marcusmueller, 2018-03-28:
OK. I'm sorry for all the distro maintainers, though, which rely on working ctest :(

noc0lour, 2018-12-13:
I'm closing this since all of the identified unit tests have their own issues (if needed) and are part of https://github.com/gnuradio/gnuradio/projects/3

closed 2018-12-13 by noc0lour