gnuradio/gnuradio #1263: grc: Add hide option to GRC Parameter

mhostetter, 2017-04-05:
This is a feature request.

I think it would be nice to add a "Hide" field in the Parameter block beneath "Short ID". This way certain fields don't have to appear on a hierarchical block. Hide could be an enum with options `None`, `Part`, or `All` and maybe default to `None`.

[image]

What do you think @skoslowski?

skoslowski, 2017-04-06:
I think, the the "Part" option is a great idea. I not sure I understand the use-case for 'All'. Why not use a variable instead of a hidden parameter? Anyway, this would have to be an enum entry.

If you want to go ahead and implement this, I'll be looking forward to your PR =)

kartikp1995, 2017-04-06:
I can work on this. If @mhostetter, hasn't started yet. 

I believe we need to update in [this file](https://github.com/gnuradio/gnuradio/blob/master/grc/core/generator/Generator.py#L291). Am I correct?

Once I get confirmation from @mhostetter, I will make the PR asap.

kartikp1995, 2017-04-06:
Added PR #1265

Kindly review it.

mhostetter, 2017-04-06:
@skoslowski you're right. The `All` option doesn't make any sense.

@kartikp1995 thanks! I would've submitted a PR, but I'm less familiar with where all the code lives in GRC land.

skoslowski, 2017-04-06:
Glad everybody's happy, but, @kartikp1995, you did indeed jump the gun a little...
@mhostetter, you're welcome to comment on or review the pr

mhostetter, 2017-04-06:
I'm building and testing the code right now.

skoslowski, 2017-04-06:
FYI, with gr installed you can run grc straight from the source tree.

kartikp1995, 2017-04-06:
@skoslowski Sorry for rushing up. I will wait for the signal next time. :)

mhostetter, 2017-04-06:
@skoslowski thanks for the tip!

mhostetter, 2017-04-19:
#1265 is now merged. Thanks @kartikp1995!

closed 2017-04-19 by mhostetter