gnuradio/gnuradio #1267: gtqui: Num Rows disappears with Byte Message type

mhostetter, 2017-04-07:
When the block is created with the `Byte` type, both `Num. Rows` and `Num. Cols` are configurable.
[image]

When switching to `Byte Message`, `Num. Rows` disappears.
[image]

I learned about this when immediately switching to `Byte Message`. After I configured then the block it complained about `Num. Rols` being invalid.

I will submit a PR for this.

mhostetter, 2017-04-07:
I just realized usage of the `Byte Message` option only prints 1 row at a time, which is different from what I expected. I achieve my desired usage by putting a PDU to Tagged Stream block between my PDU source and a `Byte` type Time Raster Sink.

Maybe this block error could be avoid if the default `Num. Rows` was `1`. Then there wouldn't be a `<check>` error.

marcusmueller, 2023-10-24:
Seems to be fixed in main

closed 2023-10-24 by marcusmueller