gnuradio/gnuradio #1271: CI: to-do list

noc0lour, 2017-04-10:
Expand test matrix:

## per PR
- [ ] next: ubuntu 16.04 qt5 
- [ ] python3: ubuntu 16.04 py2
- [ ] codecov.io

## on push:
- [ ] all PR tests (minus mergeability)

## weekly:
- [ ] all PR tests (minus mergeability)
- [ ] coverity (on master)

In general: 
specify DESIRED_QT_VERSION

jmcorgan, 2017-04-10:
For Ubuntu 16.04, next branch, QT4, you need to specify -DDESIRED_QT_VERSION=4, and use an builder image with QT4 installed.

For Ubuntu 16.04, next branch, QT5, you need to use an builder image with QT5 installed.

jmcorgan, 2017-04-10:
We need to have the deps installed for building the docs and turn that on in all the builders.

kartikp1995, 2017-04-10:
Not sure if this is the right place but add "setting up SSL certificate for `ci.gnuradio.org`" in checklist. :)

noc0lour, 2017-04-16:
@kartikp1995 look at this shiny SSL everywhere (:

noc0lour, 2018-04-07:
Move CI todo list to the CI repo: https://github.com/gnuradio/gnuradio-buildbot/issues

closed 2018-04-07 by noc0lour