gnuradio/gnuradio #1279: ATSC example file in Simulation  gnuradio is not work?

stackprogramer, 2017-04-15:
there is in ATSC file example file gnuradio-master/gr-dtv/examples:

> file_atsc_tx.grc
> file_atsc_rx.grc

when i test ATSC in simulation gnuradio it can not create any output mpeg ts file............
i even set file source complex and i delete ISShortTOcomplexBlock.........it has not any output
why simulation ATSC has problem???
i soon share video on youtube.
 thank you
best regards stackprogramer

stackprogramer, 2017-04-15:
shared video youtube problem
https://www.youtube.com/watch?v=SI4xCfmpwNM

drmpeg, 2017-04-15:
You have to match the sample rates. In file_atsc_rx.grc, change the sample_rate variable from 6.25e6 to 4500000.0 / 286 * 684.

Also, as you've already discovered, delete or bypass the ISHort to Complex block and change the File Source block Output Type from Short to Complex.

The receiver takes a while to synchronize, so the first part (maybe 5000 or so TS packets, or about 1 megabyte) of the output file will not match the input file.

drmpeg, 2017-04-15:
The test Transport Stream for file_atsc_tx.grc is available here.

http://www.w6rz.net/advatsc.ts

As described in the README.atsc file.

stackprogramer, 2017-04-16:
hi, thanks ,but with your sample rate 4500000.0 / 286 * 684. and  video file example same before we can not simulate i soon share video........

stackprogramer, 2017-04-17:
sharing video ,in debian os jessie , gnuradoi 3.7.10 we have problems with atsc.ts file that you was shared best regards stackprogramer

https://www.youtube.com/watch?v=X0cDHiFW050

jmcorgan, 2017-04-17:
The issues system here is for reporting bugs; I think your issue more one of support.  Please take this to the gnuradio mailing list for help there.

closed 2017-04-17 by jmcorgan

drmpeg, 2017-04-21:
There was a bug/regression in the PFB arbitrary resampler that was affecting the ATSC receiver in GNU Radio versions from 3.7.6 to 3.7.10.1.

You'll have to upgrade to the 3.7.10.2 or 3.7.11 release where the bug is fixed by this commit.

https://github.com/gnuradio/gnuradio/commit/915f1b02408a6a4f2ce2ffb2d9336643745aa939

stackprogramer, 2017-04-21:
@drmpeg  thanks very much, i upgraded to gnuradio 3.7.12, (before i had gnuradio 3.7.1) now it work like charm. 
best regards stackprogramer