gnuradio/gnuradio #1280: Found the ATSC overrun source - python ATSC RX Filter

ghostop14, 2017-04-16:
I've been working on seeing if I could get the ATSC pipeline to work straight from hardware to file without any overruns so I broke the hierarchical blocks into the subcomponents and started going through them one at a time.  Turns out the 1 block that was causing the biggest hit was the ATSC RX Filter block.  Looks like it was still in Python.  I ported it to C++ here https://github.com/ghostop14/gr-atsc2 and the overruns immediately stopped.  Would it be possible to take this code and incorporate it into the main GNURadio code to replace the python version?  And possibly port the other ATSC python block to C++ for an even better performance increase?

Thanks!

jmcorgan, 2017-04-16:
The ATSC RX Filter block is a Python hierarchical block.  This is a construct in GNU Radio where underlying blocks can be made to appear as a single block. However, the dataflow is still handled in the work functions of the underlying blocks; there is no DSP happening in Python.

In fact, in this hierarchical block, it is simply used as a way of forming the filter taps of a polyphase filter/resampler block, so the single degree of freedom (resampling ratio, or samples per ATSC symbol) can control the various parameters in the filter so the user doesn't have to calculate them.

Hierarchical blocks can also be written in C++, as you've done, but their performance is identical to Python hierarchical blocks, as it is actually the same lower-level pfb resampler doing the work.  There is no processing happening in Python on a per-sample basis, only initial configuration when constructing the block.

If you are seeing a difference, I would look to something else that has changed in your two scenarios. 

Since the ATSC RX Filter is the first block in the DSP chain, it is exposed to the fastest data rate coming from the RF source, and has to implement both FIR filtering and resampling to an even higher data rate.  It is usually either the highest or second highest CPU consumer in the flowgraph (the other being the Viterbi decoder(s) on the digital side.)  Its load increases linearly with the resample rate, so my first suggestion would be to look to see if the rates are different between the native Python hierarchical block and your reimplementation in C++.

ghostop14, 2017-04-17:
Hmmm, thanks for the info on the python/c++ hierarchical blocks.  In that case I'll keep looking.

closed 2017-04-17 by ghostop14