gnuradio/gnuradio #1288: FileSink should have an option to fail if file already exists

giladbeeri, 2017-04-19:
FileSink has 2 options: append or overwrite.
Sometimes users forget to change the filename but they don't intend to either lose data (overwrite) or modify it (append).
An option to configure FileSink to throw an exception if the file already exists will be useful.

marcusmueller, 2023-12-20:
Valid, sensible Feature Request

marcusmueller, 2025-03-09:
I marked this ["good first issue"](https://github.com/gnuradio/gnuradio/issues?q=state%3Aopen%20label%3A%22good%20first%20issue%22).

This is for new contributors who

- know C++
- can locally compile GNU Radio
- would like to try out writing a Python or C++ unit test (there's already unit tests, you just need to extend these) that checks whether, if a "fail if file exists" flag is set, the constructor of the file sink block throws an exception
- know enough git to make a pull request and use `--signoff` when they `git commit`

closed 2026-08-18 by marcusmueller