gnuradio/gnuradio #1302: Msg input ports not connected to anything are not exposed to Ctrlport

777arc, 2017-04-26:
While debugging why the example flowgraph **simple_copy.grc** (located in gr-blocks/examples/ctrlport/) wouldn't work I discovered that message input ports that aren't connected to any other block do not become exposed and accessible via ControlPort, so commands like postMessage() won't reach them.  So to get **simple_copy.grc** to work I had to connect a block containing a message output port to the Copy block, which allowed the simple_copy_controller.py to actually enable and disable the block through ControlPort.  Otherwise this example just doesn't work the way it's intended.   

I assume that when this example was written, msg ports without anything connected would still exist and be exposed to ControlPort, but something must have changed since then.  So I think the two options to solve this issue are:

1) Change the flowgraph generator so unconnected message inputs are still active
2) Create a block that has an output message port but doesn't actually produce any messages, so it can be tacked on to this example, and any other flowgraphs that won't work due to this issue.  Here is an example of what I mean:
```python
# Block that doesn't do anything, used to get a msg input port on another block exposed to ControlPort
from gnuradio import gr
import pmt
class blk(gr.basic_block):
    def __init__(self): 
        gr.basic_block.__init__(self,name='Stupid Block',in_sig=[],out_sig=[])
        self.message_port_register_out(pmt.intern("fake_output"))
```

mhostetter, 2017-04-26:
I think this is related to what I saw in #1239.

777arc, 2017-04-26:
I don't quite see how it's related to #1239

mhostetter, 2017-04-27:
What I experienced in #1239 was a source block with an optional message output. When the block was instantiated in GRC but the message output was not connected, the block didn't seem to run properly. I suspected that because the block wasn't connected to the flowgraph via a `msg_connect()` call, that the block wasn't being "included" in the top block somehow, but I wasn't sure.

This seemed similar to me when you said:
> I assume that when this example was written, msg ports without anything connected would still exist and be exposed to ControlPort, but something must have changed since then.

777arc, 2017-04-27:
Oh ok.  Well with my issue, the block was definitely added to top block and functioned like normal, except for the message input port not working.

mhostetter, 2017-04-27:
Yeah, they may not be related at all, but my ears perked up when I read the comment I quoted.

bastibl, 2017-04-27:
I don't think that they are related. I guess since this [commit](https://github.com/gnuradio/gnuradio/commit/080bbf3dfcef968a2141714852bf691ba442bb2f) `setup_rpc` is not called for all blocks. We could try something like [this](https://github.com/bastibl/gnuradio/commit/52eab1718a345fafbe82fa1166c57f8caad0b506), but I didn't fully understand `flatten_aux`. Maybe you could have a look?

mrjacobagilbert, 2017-10-01:
I just ran into something simiarl.

As I recall, prior to my PR, setup_rpc() was only called for blocks with connected streaming ports, and was updated to call setup_rpc() for all blocks with connected ports (message or streaming).

If this is still an open issue I will take a look since it looks like I might have broken this.

777arc, 2017-10-01:
awesome thanks!

mrjacobagilbert, 2017-10-02:
Yep, that PR fixed performance counters for message port only blocks but broke how setup_rpc() is called for blocks without message connections. Wow. I'll look into it and get a PR in to address this.

mormj, 2021-10-04:
@jacobagilbert / @777arc - Appears the two PRs referenced here fix this issue?  Can this be closed now?

marcusmueller, 2023-12-20:
Seems like it

closed 2023-12-20 by marcusmueller