gnuradio/gnuradio #1309: gnuradio.analog.am_demod assumes input is normalized

kpreid, 2017-05-10:
`gnuradio.analog.am_demod` says that

> This block demodulates a band-limited, complex down-converted AM
> channel into the the original baseband signal, applying low pass
> filtering to the audio output. It produces a float stream in the
> range [-1.0, +1.0].

However, if the input AM signal's carrier is not of magnitude 1.0, then:

* the output will have a DC offset (because the block assumes that subtracting 1 will remove the carrier), and
* the output will not in fact be in the range [-1.0, +1.0], but a larger or smaller range.

Either the `am_demod` block should be changed to contain an AGC block, or the requirement on its input should be documented.

jmcorgan, 2017-05-10:
Given there is now gr::filter::dc_blocker_ff (which didn't exist at the time that AM demod hier block was written), it would be straightforward to replace the DC-reject subtractor. This would make the block suitable for non-normalized input, while not changing its behavior for normalized signals. 

Would you be interested in doing this and submitting a pull request?

kpreid, 2017-05-11:
> Given there is now gr::filter::dc_blocker_ff (which didn't exist at the time that AM demod hier block was written), it would be straightforward to replace the DC-reject subtractor. This would make the block suitable for non-normalized input, while not changing its behavior for normalized signals.

DC blocking is only a partial solution, because the output audio signal's magnitude will still vary depending on the RF signal strength, which it should not (at least, that is my understanding of how an audio demodulator should operate). An AGC (which is just another control loop for a different parameter, anyway) will address both issues.

> Would you be interested in doing this and submitting a pull request?

I've already not gotten to something that I care about a lot more than this, so it's unlikely I'll get around to it. Sorry.

(I don't consider this a high priority; I simply noticed it while [giving some advice](https://lists.gnu.org/archive/html/discuss-gnuradio/2017-05/msg00083.html) and want to make sure it was written down.)

closed 2017-05-11 by jmcorgan