gnuradio/gnuradio #1314: Throtle needed in a system with a audio sink

fperal, 2017-05-17:
[image]

It works (sound is played ok) with samples/symbol=8 or samples/symbols=2, but ONLY if the throthle is enabled. If the throthle is bypassed or removed the sound is intermittent.

marcusmueller, 2017-05-18:
admins: assign this to me, please. 

my current guess is that I need to take a close look at what message queuing happens inside the Packet Encoder and the Demod block. There's something :fish:y happening here.

marcusmueller, 2017-05-18:
Adding the .grc from the [email thread](http://lists.gnu.org/archive/html/discuss-gnuradio/2017-05/msg00105.html)
[image]

https://gist.github.com/marcusmueller/1ff825bea30888efab07283a1deea0f1

jmcorgan, 2017-05-18:
The packet encoder hierarchical block has been deprecated in the 3.7 series, and it's on the list of things to reimplement for the gr-digital cleanup in 3.8.  So, if the issue is in the underlying GNU Radio stuff, or something simple in the hierarchical block, we should fix right away. If it is complicated fix or reimplementation, it probably isn't worth writing code that will get removed later.

LaughingCabbage, 2017-07-13:
My theory here is that by removing the throttle you are allowing GRC to consume the .wav file at an arbitrary rate that likely exceeds the desired sample rate. Thus, the samples arriving at the audio sink are not truly at 44.1kHz. Thoughts?

mbr0wn, 2017-07-14:
That shouldn't matter though. Samples will never arrive an audio source at the correct sampling rate, but it should then properly backpressure. The rate at which the .wav file is consumed won't make a difference.

Still, any of the blocks in this flow graph could be the culprit. Or it's some scheduler randomness bug, which is bypassed with a throttle because all the buffers will always empty.

marcusmueller, 2017-07-14:
Agreeing with @mbr0wn, GNU Radio is a DSP system, where it's irrelevant how fast samples are produced or processed - the end result should be the same.

Haven't found out reason for strangeness yet.

jmcorgan, 2017-07-14:
Also, please ensure 'ok to block' is true in the audio sink config.

LaughingCabbage, 2017-07-14:
The current main branch's packet decoder has an open bug that I found ( #1370  ) which discards the final packet from a transmitted message if it does not satisfy the hard-coded DEFAULT_PAYLOAD_LENGTH. Thus, if a message is consumed by the packet decoder and the message size % payload length is not 0 then the packet will be discarded and the decoder will move on to consume the next available message from the queue. This data loss could be the source of your intermittent sound with a maximum loss of DEFAULT_PAYLOAD_LENGTH-1 bytes (or 511 bytes) at the end of any given message.

marcusmueller, 2021-03-22:
packet_encoder has been removed; I'll go ahead and close this issue.. If it appears in other form, please don't hesitate to open a new one and reference this!

closed 2021-03-22 by marcusmueller