gnuradio/gnuradio #1316: gnuradio-3.7.11 compilation fails with gcc-7

yarda, 2017-05-24:
I was upgrading gnuradio in fedora rawhide. With the gnuradio default -std=c++98 it failed:
```cd /builddir/build/BUILD/gnuradio-3.7.11/build/gnuradio-runtime/lib && /usr/bin/c++  -DALIGNED_MALLOC=0 -DENABLE_GR_LOG -DGR_CTRLPORT -DGR_PERFORMANCE_COUNTERS -DHAVE_ARPA_INET_H -DHAVE_BYTESWAP_H -DHAVE_COSF -DHAVE_GETPAGESIZE -DHAVE_GETTIMEOFDAY -DHAVE_LINUX_PPDEV_H -DHAVE_MALLOC_H -DHAVE_MMAP -DHAVE_NANOSLEEP -DHAVE_NETDB_H -DHAVE_NETINET_IN_H -DHAVE_POSIX_MEMALIGN -DHAVE_PTHREAD_SETSCHEDPARAM -DHAVE_PTHREAD_SIGMASK -DHAVE_SCHED_SETSCHEDULER -DHAVE_SELECT -DHAVE_SHM_OPEN -DHAVE_SIGACTION -DHAVE_SIGNAL_H -DHAVE_SINCOS -DHAVE_SINCOSF -DHAVE_SINF -DHAVE_SNPRINTF -DHAVE_SYSCONF -DHAVE_SYS_IPC_H -DHAVE_SYS_MMAN_H -DHAVE_SYS_RESOURCE_H -DHAVE_SYS_SELECT_H -DHAVE_SYS_SHM_H -DHAVE_SYS_SOCKET_H -DHAVE_SYS_TIME_H -DHAVE_SYS_TYPES_H -DHAVE_UNISTD_H -DHAVE_VALLOC -DTRY_SHM_VMCIRCBUF -Dtest_gnuradio_runtime_EXPORTS -I/builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/math -I/builddir/build/BUILD/gnuradio-3.7.11/build -I/builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/include -I/builddir/build/BUILD/gnuradio-3.7.11/build/gnuradio-runtime/include -I/builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib -I/builddir/build/BUILD/gnuradio-3.7.11/build/gnuradio-runtime/lib/../include -I/builddir/build/BUILD/gnuradio-3.7.11/volk/include -I/builddir/build/BUILD/gnuradio-3.7.11/build/volk/include -I/builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/controlport -I/builddir/build/BUILD/gnuradio-3.7.11/build/gnuradio-runtime/lib/controlport  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -std=c++98 -fvisibility=hidden -Wsign-compare -Wno-uninitialized -DNDEBUG -fPIC   -o CMakeFiles/test-gnuradio-runtime.dir/math/qa_fxpt_nco.cc.o -c /builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/math/qa_fxpt_nco.cc
In file included from /builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/math/qa_fxpt.h:26:0,
                 from /builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/math/qa_fxpt.cc:27:
/builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/math/qa_fxpt.h: In static member function 'static CppUnit::TestSuite* qa_fxpt::suite()':
/builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/math/qa_fxpt.h:36:3: error: 'unique_ptr' is not a member of 'std'
   CPPUNIT_TEST_SUITE_END();
   ^
```

And:
```In file included from /usr/include/c++/7/type_traits:35:0,
                 from /usr/include/cppunit/tools/StringHelper.h:7,
                 from /usr/include/cppunit/TestAssert.h:8,
                 from /usr/include/cppunit/TestCase.h:6,
                 from /usr/include/cppunit/TestCaller.h:5,
                 from /usr/include/cppunit/extensions/HelperMacros.h:9,
                 from /builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/pmt/qa_pmt_prims.h:27,
                 from /builddir/build/BUILD/gnuradio-3.7.11/gnuradio-runtime/lib/pmt/qa_pmt.cc:28:
/usr/include/c++/7/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
```

So I switched to gcc default -std=c++14, but it failed:
```In file included from /builddir/build/BUILD/gnuradio-3.7.11/gr-fec/include/gnuradio/fec/polar_decoder_sc.h:28:0,
                 from /builddir/build/BUILD/gnuradio-3.7.11/gr-fec/lib/polar_decoder_sc.cc:28:
/builddir/build/BUILD/gnuradio-3.7.11/gr-fec/include/gnuradio/fec/polar_decoder_common.h:71:47: error: constexpr static data member 'D_LLR_FACTOR' must have an initializer
```

I finally compiled it with -std=c++14 by using the following patch:
https://src.fedoraproject.org/cgit/rpms/gnuradio.git/tree/gnuradio-3.7.11-gcc-7-compile-fix.patch

michaelld, 2017-05-25:
CppUnit 1.14.0 requires C++11 for its API. That's the first 2 issues. The latter has been fixed in master and next already. I'll push the CppUnit fixes into a PR shortly. They are already in MacPorts for Mac OS X users.

michaelld, 2017-07-15:
"shortly" has been way too long. I'll get there soon. Sorry!

michaelld, 2017-08-14:
https://github.com/gnuradio/gnuradio/pull/1424
https://github.com/gnuradio/gnuradio/pull/1425

michaelld, 2017-08-14:
I think either 1424 or 1425 should be mergeable on next, but if not I’ll make a PR for that too — similar, yet different!

mbr0wn, 2017-10-03:
#1424 and #1425 are merged, can this issue be closed?

michaelld, 2017-10-03:
From my perspective: yes.

yarda, 2017-10-04:
Thanks, as the original requester I am closing it.

closed 2017-10-04 by yarda