gnuradio/gnuradio #1317: qtgui_time_sink can't display more than 12 inputs

777arc, 2017-05-26:
I noticed that qtgui_time_sink segfaults when I provide more than 12 inputs.  I was able to fix the issue and allow more inputs by going into TimeDomainDisplayPlot.cc and adding more QColors to the "colors" list (I added the last line to allow 3 more inputs, pushing it to 15 total):

QList<QColor> colors;
colors << QColor(Qt::blue) << QColor(Qt::red) << QColor(Qt::green)
	 << QColor(Qt::black) << QColor(Qt::cyan) << QColor(Qt::magenta)
	 << QColor(Qt::yellow) << QColor(Qt::gray) << QColor(Qt::darkRed)
	 << QColor(Qt::darkGreen) << QColor(Qt::darkBlue) << QColor(Qt::darkGray) 
         << QColor(Qt::darkGray) << QColor(Qt::darkGray) << QColor(Qt::darkGray); 

I guess those are the default colors?  Either way, we should probably allow for more than 12 inputs if desired, maybe more like 20 or 30, even if the colors are all the same.  If you're curious, the reason I am using so many damn inputs is because I only display a few at a time, but I need to be able to choose from a large number.  I've never put in a pull request before, but I am happy to if someone can point me to instructions for doing it properly.

777arc, 2017-05-26:
Also, if anyone knows how I can create my own version of this block in my OOT with the change, that would be awesome.  That way I wont need to make the change and recompile gnuradio every time I want to use my app.  I tried to just make my own time_sink_f block, but it depends on too much qtgui stuff so I feel like I would have to copy the entire gr-qtgui into my OOT to get it to work.

jmcorgan, 2017-06-07:
I'm not averse to merging this per se, but I'd rather see the limits documented and enforced, then we can consider what the limit should be.

noc0lour, 2017-10-31:
So, time sink now support 24 Connections and enforces it.

closed 2017-10-31 by noc0lour