gnuradio/gnuradio #1318: gr_modtool created OOT modules don't track dependencies correctly for swig

pwicks86, 2017-05-31:
To reproduce:

1. Create an oot module (gr_foo say)
1. Create a new C++ block via gr_modtool (bar for example)
2. Build
3. Change the C++ definition (in lib/bar_impl.* and include/foo/bar.h``) to add a new function to the block
4. Build
5. Observe that the foomod_swigPYTHON_wrap.cxx hasn't been updated to reflect the new function (alternately, if you remove a function from the block, you'll get a compile error)

The easy fix is to just `make clean` and rebuild, but it seems like cmake should be able to do better. 

I've got a local patch for this that basically just adds something like this to the CMakeLists.txt for the swig directory:

```
foreach(foomod_source_file ${foomod_sources})
    if(IS_ABSOLUTE skysafe_source_file)
        list(APPEND GR_SWIG_SOURCE_DEPS ${foomod_source_file})
    else()
        list(APPEND GR_SWIG_SOURCE_DEPS ${CMAKE_SOURCE_DIR}/lib/${foomod_source_file})
    endif()
endforeach()
```

I'm willing to submit a PR for that, but it seems like maybe that's not the best way to handle this, since it won't work in the case of changes that are confined to just the impl header files.

jmcorgan, 2017-05-31:
This is a long standing bug. I've only seen it happen when changing the include/foo/bar.h public API header; it fails to trigger a SWIG regeneration.

There should be no need to trigger a SWIG rebuild if only the impl header changes, as those aren't exposed publicly and SWIG never sees them.

Please submit a PR for review, for the gr-utils template for this.  There may be a better way to do it, but at least it will be a place to discuss that.

pwicks86, 2017-05-31:
Created #1320 for this.

pwicks86, 2017-06-07:
#1320 has been merged, closing this issue.

closed 2017-06-07 by pwicks86