gnuradio/gnuradio #1319: Total Fail...

jamieduk, 2017-05-31:
i just installed volk from git hub source because gnuradio wont install and still get this message CMake Error at CMakeLists.txt:428 (message):
  VOLK required but not found.

sort it out ffs linux winds me right up incompetent devs need to make prerequisites install automatically i have not got time to waste like this ffs!

pwicks86, 2017-05-31:
You need to check out the volk submodule. The easiest way to do this is 

```
git submodule init 
git submodule update
```

mbr0wn, 2017-06-01:
@jamieduk Please keep the tone civilized here. I'll be closing this issue because it's not actually a bug.

mbr0wn, 2017-06-01:
Also, see @pwicks86 helpful response which should resolve your issues.

closed 2017-06-01 by mbr0wn