gnuradio/gnuradio #1326: Two HackRF One Doesn't work in Gnuradio Simultaneously

stackprogramer, 2017-06-08:
In gnuradio 3.7.12 we have two hackrf one, we want to see spectrum we test in Debain jessie and Ubuntu linux all of them can not play two HackRFOne Simultaneously.
you can see video on youtube for more info....
https://www.youtube.com/watch?v=xu9qPR_QH0E
can we receive data from two hackrfone with osmocomsource Simultaneously
please help
best regards stackprogramer

ghostop14, 2017-06-08:
I don't know if this will help or not, but I did notice in your flowgraph that the sample rate is 32k.  I've personally had issues with the HackRF at low sample rates until I saw the ADC lower limit.  Michael Ossmann has a post on the HackRF site for "Tips and Tricks" (https://github.com/mossmann/hackrf/wiki/Tips-and-Tricks) that talks about the MAX5864 ADC not being specified to operate below 8 MHz.  32K is REALLY low.  Don't know if that's contributing to your problem or not.  I've successfully done a HackRF and an Airspy with synchronized clocks in the same flowgraph to frequency sinks and it's worked well.  I'd also hit a HackRF firmware and library update just to rule it out.  There were a number of fixes earlier this year in February that require a firmware update to correct.

stackprogramer, 2017-06-09:
@ghostop14  i before update firmware...
but for low sample rate i before transmited and received FM in gnuradio with sample rate 50k,
we have not any problem....
i will examine more and share result..
thank you

n-west, 2017-06-09:
This doesn't seem like a gnuradio issue or bug. It's likely either a bug with gr-osmosdr or with hackrf. Additionally, if you just need tech support the mailing lists are the better and preferred place to go.

jmcorgan, 2017-06-09:
Agree.

closed 2017-06-09 by jmcorgan

stackprogramer, 2017-06-15:
@ghostop14 Hi,today i test hackrf one sample set to 1M,now it works thanks for attention
best regards stackprogramer