gnuradio/gnuradio #1329: Illegal Instruction on OS X

mehdideveloper, 2017-06-11:
Hi
I compiled 3.7.11 on OS X 10.12.5 and when I try to run grc, I get this error:

> <<< Welcome to GNU Radio Companion 3.7.11 >>>
>
>Block paths:
>	/Users/mehdi/.grc_gnuradio
>	/usr/local/share/gnuradio/grc/blocks
>Illegal instruction: 4

And it stops. 
Any ideas?
I haven't been able to run GNURadio on OS X after updating to 10.12 (I've tested brew, but it does not support grc because of Qt4 deprecation; I don't have port)

michaelld, 2017-07-15:
Could be many possible issues. How did you install GNU Radio in the first place, if not via port or brew? How did you install the dependencies?

mehdideveloper, 2017-07-15:
Well, I finally managed to resolve this, and in the process, I tried endless things to fix it (which I don't even remember to list)
I've tried both brew and compiling from source (I don't use port)
At last, I removed every dependency that I suspected were related to the problem (like swig) and manually deleted remaining header and lib files, and compiled the latest gnuradio from source (and I had to edit the cmake files with "ccmake .." command to change all library directories to point to the ones installed via brew (like Python, Swig, boost, etc
Even after it, I had segfaults with "gr_filter_design" which I fixed (after lots of trial and error) by compiling PyQwt from source (and  deleting old ".so" files from the system)
So I think it was caused by library version mismatches (for example in the "gr_filter_design" case, my "Qwt.so" was pointing to "qt 4.8.6" while I had "qt 4.8.7" installed on my system)

Anyway, thanks for the reply.

closed 2017-07-15 by mehdideveloper

michaelld, 2017-07-15:
Yes: Probably library version mismatches, given the way you installed GR and its dependencies. I'm glad you got it all working, well enough. Great work!

michaelld, 2017-07-15:
@jmcorgan Please close this issue.