gnuradio/gnuradio #1337: multiply_matrix_XX: can't simply use magic value in place of `enum`, C++ doesn't allow that

marcusmueller, 2017-06-20:
```
[793/2135] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/multiply_matrix_cc_impl.cc.o
../gr-blocks/lib/multiply_matrix_cc_impl.cc:184:15: warning: comparison of constant 999 with expression of type 'gr::block::tag_propagation_policy_t' is always false [-Wtautological-constant-out-of-range-compare]
      if (tpp == TPP_SELECT_BY_MATRIX) {
          ~~~ ^  ~~~~~~~~~~~~~~~~~~~~
1 warning generated.
```

Simply because you might in practice be able to cast the `tag_propagation_policy_t` argument to `set_tag_propagation_policy` to an `int` doesn't mean you can just set it to `999`. That's not how `enum`s work.

Especially, the underlying data type used to carry the `enum` is totally up to the compiler to choose – and it might just as well be picking an 8bit type for this, for which `999` is impossible.

So, I'm not sure how to proceed. Options:

1. extend `tag_propagation_policy_t` to include `TPP_SELECT_BY_MATRIX`. That demonstrates that we can't keep the number of possible TPPs limited.
2. extend `tag_propagation_policy_t` to include `TPP_CUSTOM`, and change `multiply_matrix` + docs to react to that
3. ???

gigabyte1023, 2017-06-21:
i need help

mbr0wn, 2017-06-22:
We actually had a discussion about this when this block was first merged. The `TPP_CUSTOM` solution was the one that I preferred at the time.

marcusmueller, 2017-06-22:
Ok, then I'll just go ahead and implement that – Thanks for the feedback!

mbr0wn, 2017-06-22:
I think @jmcorgan should weigh in here before you go on a coding spree.

jmcorgan, 2017-06-23:
I think I agree with the TPP_CUSTOM solution myself.

marcusmueller, 2017-06-25:
@jmcorgan (hopefully) ready for maint-merge (#1341)

noc0lour, 2017-11-01:
@marcusmueller This is done, I guess?

closed 2017-11-01 by noc0lour

marcusmueller, 2017-11-01:
@noc0lour as far as I can see, yes, the fix has been merged into maint, which, I think, in the GNU Radio dev model, means it'll end up (or already has ended up) in master and next.