gnuradio/gnuradio #1338: GMSK Modulation Block Bugs

estradjm, 2017-06-21:
There are two GMSK modulation blocks, GMSK Mod and GMSK Modulation, of which both do not work correctly. GMSK Mod block is definitely not working correctly. 

Flow graphs and screen shots available here:
https://lists.gnu.org/archive/html/discuss-gnuradio/2017-06/msg00209.html

gigabyte1023, 2017-06-21:
i need help

awalls-cx18, 2017-07-21:
Please read the python source code for these blocks to learn how to format their input streams:
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/python/digital/cpm.py
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/python/digital/gmsk.py

From the png image of your flowgraph, it appears to me that you may not have formed the input stream in a manner that the blocks require.

mormj, 2021-03-24:
The awalls guidance leads me to close this one (and due to inactivity)

closed 2021-03-24 by mormj