gnuradio/gnuradio #1345: Ubuntu 14.04.5 i686 (32bits) does not show right block list on gnuradio-companion

braselectron, 2017-06-25:
After a reinstall gqrx and grc using ppa repository the left list of blocks on grc window just does not show, so there is no way to make/edit my projects.

I can open a project that I know that works and it work fine, but can't add blocks (but I can remove).

[image]
uname -a:
3.13.0-121-generic #170-Ubuntu SMP Wed Jun 14 09:05:46 UTC 2017 i686 i686 i686 GNU/Linux

repository 
     ppa:bladerf/bladerf
     ppa:ettusresearch/uhd
     ppa:myriadrf/drivers
     ppa:myriadrf/gnuradio
     ppa:gqrx/gqrx-sdr
gqrx works.

Package: 
    libvolk1-bin  Versions: 1.3-1build1~ubuntu14.04.1myriadrf1
    gnuradio Versions: 3.7.10.1-2~ubuntu14.04.1~ppa1
    gqrx-sdr  Versions: 2.6.1-gqrx1~trusty
    rtl-sdr Versions: 0.5.3-myriadrf1~trusty
    gnuradio-dev  Versions: 3.7.10.1-2~ubuntu14.04.1~ppa1

Please help.

kartikp1995, 2017-06-25:
Can you find the XML files of the blocks in the block paths mentioned at the bottom of GRC window?

braselectron, 2017-06-25:
Yes

skoslowski, 2017-06-25:
Might be, the panel is just hidden. Check the application menu.

braselectron, 2017-06-25:
@skoslowski unfortunately it is not a hidden panel, I did check for that first.  Or could you explain where?

braselectron, 2017-06-25:
I did this, to check if it is a commit (update) problem:

I removed (purge) gqrx, gnuradio*, touble checked builds make (make uninstall), removed the repository ppa's installed and returned ubuntu main/standard repository and used software center to install gnuradio GNU Radio Companion 3.7.2.1 (which is the one officially available).

Then I installed the gqrx-sdr from standard repository too.

now I have the panel back!  

[image]

So I suspect this is a commit bug.

braselectron, 2017-06-26:
OK  So after sending my Sunday on this, I found a workaround.

First check where are the grc.conf files on your system
rename all that are not in the predictable locations, which are:
```
~/.gnuradio/
/etc/gnuradio/conf.d/
```

Then go to

```
/etc/gnuradio/conf.d/
gksu gedit grc.conf
```

copy the lines (or the paths in your system):

```
[grc]
global_blocks_path = /usr/share/gnuradio/grc/blocks:/usr/local/share/gnuradio/grc/blocks
local_blocks_path =
```

close file

```
cd ~/.gnuradio
gedit grc.conf
```

paste lines after [main] (mush have one space after last group of lines separating groups)

save ~/.gnuradio/grc.conf

close file

open gnuradio-companion

block list returned.  Hurrah!

close/exit app

open using terminal (and you will see errors) again you get a error (crash)

open conf files again and remove the pasted lines and any ['x'] that will show around some variables

save files

That is it!  Finish bug workaround.

closed 2017-06-27 by jmcorgan

braselectron, 2017-06-28:
@jmcorgan please note, that I did not close the topic for the following reasons:

1. because the "workaround" solved the PPA repository re-installation update "bug" but,
1.1. not sure if this is a "bug" or back-port issue to Trusty Tahr or 32bits
2. and also the fact that I tried to build this project and for my surprise, it "abended" (I was not able to build it for ubuntu 14.04.5 i686 any more - I did it for the last three years with no major problem).

So please consider this too, and we might need to open new issue report for the build issue.

If you need to test the build issue I can help with a CPU~Dual core Intel Core2 CPU 6300 (-MCP-) clocked at 1866.520 Mhz Kernel~3.13.0-121-generic i686 Up~22 min Mem 8GB HDD~500.1GB and also a Intel NUC6i5SYH (i5), and Lenovo i7 and Rpi model 2 and 3 (running Raspbian Linux 4.1.19+)

Best regards.