gnuradio/gnuradio #1346: gr-vocoder/codec2 code hygiene: dumps struct with float member, reads ints

marcusmueller, 2017-06-25:
My compilers have a hard time, because this 
```
const struct lsp_codebook lsp_cbjnd[] = {
… ,
  /* /home/marcus/src/gnuradio/gr-vocoder/lib/codec2/codebook/lspjnd5-10.txt */
  {
    6,
    11.7181,
    3369,
    codes4
  },
…
```
initializes the second field of an `lsp_codebook` struct with the floating point literal 11.7181, but in fact, that gets downcasted (and truncated) to an int (in codec2/defines.h):

```
struct lsp_codebook {
    int			k;        /* dimension of vector	*/
    int			log2m;    /* number of bits in m	*/
    int			m;        /* elements in codebook	*/
    const float	*	cb;	  /* The elements		*/
};
```

Now, I'm not sure we should be letting the compiler convert 11.7 to 11 bits per `m`…

I looked into updating Codec2 from rowetel, but the codebook generator still has the same "write floating point string, have int in structure" issue.

Since it's a legitimate thing for a compiler to complain about, and I do have my computational doubts about what Codec2 does being correct, I'm opening this issue to remind us of it, should Codec2 cause trouble beyond a compiler warning.

jmcorgan, 2017-06-27:
Codec2 is finally getting packaged in various places; for 3.8 we've already removed this code and link to a system installed version. I wouldn't waste your time on this.

marcusmueller, 2017-06-27:
In that case: Closing this.

closed 2017-06-27 by marcusmueller