gnuradio/gnuradio #1347: Python crash on Mac OS X 10.12.5

cn0xroot, 2017-06-27:
```
Process:               Python [16490]
Path:                  /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               2.7.13 (2.7.13)
Code Type:             X86-64 (Native)
Parent Process:        Python [16476]
Responsible:           Python [16490]
User ID:               501

Date/Time:             2017-06-27 12:27:15.994 +0800
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
Anonymous UUID:        902D2FEB-357F-7C2F-6BC0-D460C494C2FC

Time Awake Since Boot: 9600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000003802000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x3802000000:
    Memory Tag 251         000000014491b000-000000014491e000 [   12K] rw-/rwx SM=PRV  
--> 
    STACK GUARD            000070000eec5000-000070000eec6000 [    4K] ---/rwx SM=NUL  stack guard for thread 1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libgnuradio-blocks.3.7.11.dylib	0x000000011337d7f6 boost::asio::detail::task_io_service::init_task() + 134
1   libgnuradio-blocks.3.7.11.dylib	0x0000000113362477 boost::asio::io_service::service* boost::asio::detail::service_registry::create<boost::asio::datagram_socket_service<boost::asio::ip::udp> >(boost::asio::io_service&) + 105
2   libuhd.003.dylib              	0x0000000119fe1bbf boost::asio::detail::service_registry::do_use_service(boost::asio::io_service::service::key const&, boost::asio::io_service::service* (*)(boost::asio::io_service&)) + 127
3   libuhd.003.dylib              	0x000000011a1ce72c udp_simple_impl::udp_simple_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool, bool) + 844
4   libuhd.003.dylib              	0x000000011a1ce31b uhd::transport::udp_simple::make_broadcast(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 55
5   libuhd.003.dylib              	0x000000011a04daff uhd::usrp::e300::e300_find(uhd::device_addr_t const&) + 1717
6   libuhd.003.dylib              	0x000000011a04da1d uhd::usrp::e300::e300_find(uhd::device_addr_t const&) + 1491
7   libuhd.003.dylib              	0x000000011a120591 boost::detail::function::function_invoker1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) + 17
8   libuhd.003.dylib              	0x000000011a1d8d4c boost::function1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::operator()(uhd::device_addr_t const&) const + 34
9   libuhd.003.dylib              	0x000000011a1d8b7d uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) + 121
10  libgnuradio-osmosdr.0.1.5git.dylib	0x0000000119b95b42 uhd_source_c::get_devices() + 120
11  libgnuradio-osmosdr.0.1.5git.dylib	0x0000000119b66c7d source_impl::source_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 3379
12  libgnuradio-osmosdr.0.1.5git.dylib	0x0000000119b65eb9 osmosdr::source::make(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 43
13  _osmosdr_swig.so              	0x0000000119ab5aa8 _wrap_source_make(_object*, _object*, _object*) + 201
14  org.python.python             	0x000000010d0ead43 PyEval_EvalFrameEx + 32947
15  org.python.python             	0x000000010d0e2947 PyEval_EvalCodeEx + 2119
16  org.python.python             	0x000000010d0eed46 fast_function + 118
17  org.python.python             	0x000000010d0e9e87 PyEval_EvalFrameEx + 29175
18  org.python.python             	0x000000010d0e2947 PyEval_EvalCodeEx + 2119
19  org.python.python             	0x000000010d0665bb function_call + 363
20  org.python.python             	0x000000010d03f203 PyObject_Call + 99
21  org.python.python             	0x000000010d04d3a6 instancemethod_call + 182
22  org.python.python             	0x000000010d03f203 PyObject_Call + 99
23  org.python.python             	0x000000010d09e90f slot_tp_init + 175
24  org.python.python             	0x000000010d09aadb type_call + 347
25  org.python.python             	0x000000010d03f203 PyObject_Call + 99
26  org.python.python             	0x000000010d0ea19d PyEval_EvalFrameEx + 29965
27  org.python.python             	0x000000010d0e2947 PyEval_EvalCodeEx + 2119
28  org.python.python             	0x000000010d0eed46 fast_function + 118
29  org.python.python             	0x000000010d0e9e87 PyEval_EvalFrameEx + 29175
30  org.python.python             	0x000000010d0e2947 PyEval_EvalCodeEx + 2119
31  org.python.python             	0x000000010d0e20f6 PyEval_EvalCode + 54
32  org.python.python             	0x000000010d111824 PyRun_FileExFlags + 164
33  org.python.python             	0x000000010d11135e PyRun_SimpleFileExFlags + 702
34  org.python.python             	0x000000010d127b1d Py_Main + 2925
[... 358 lines cut ...]
    0x7fffabcd3000 -     0x7fffabcdcff3  libsystem_notify.dylib (165.20.1) <B8160190-A069-3B3A-BDF6-2AA408221FAE> /usr/lib/system/libsystem_notify.dylib
    0x7fffabcdd000 -     0x7fffabce5fe7  libsystem_platform.dylib (126.50.8) <897462FD-B318-321B-A554-E61982630F7E> /usr/lib/system/libsystem_platform.dylib
    0x7fffabce6000 -     0x7fffabcf0ff7  libsystem_pthread.dylib (218.60.3) <B8FB5E20-3295-39E2-B5EB-B464D1D4B104> /usr/lib/system/libsystem_pthread.dylib
    0x7fffabcf1000 -     0x7fffabcf4ff7  libsystem_sandbox.dylib (592.60.1) <DC780631-BD23-36B1-9376-668619E18D25> /usr/lib/system/libsystem_sandbox.dylib
    0x7fffabcf5000 -     0x7fffabcf6ff3  libsystem_secinit.dylib (24.50.4) <F78B847B-3565-3E4B-98A6-F7AD40392E2D> /usr/lib/system/libsystem_secinit.dylib
    0x7fffabcf7000 -     0x7fffabcfeffb  libsystem_symptoms.dylib (532.50.47) <3390E07C-C1CE-348F-ADBD-2C5440B45EAA> /usr/lib/system/libsystem_symptoms.dylib
    0x7fffabcff000 -     0x7fffabd12ff7  libsystem_trace.dylib (518.60.2) <6B145B10-5874-3E89-90CD-D370DB475BA1> /usr/lib/system/libsystem_trace.dylib
    0x7fffabd13000 -     0x7fffabd18ffb  libunwind.dylib (35.3) <3D50D8A8-C460-334D-A519-2DA841102C6B> /usr/lib/system/libunwind.dylib
    0x7fffabd19000 -     0x7fffabd42ff7  libxpc.dylib (972.60.2) <1C9AF716-69DF-359F-85E9-7DFDE362F9A2> /usr/lib/system/libxpc.dylib

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 1
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 6426
    thread_create: 0
    thread_set_state: 0

```

michaelld, 2017-06-27:
Lucky me!

michaelld, 2017-06-27:
@cn0xroot : What command line are you using that generated this error?

michaelld, 2017-06-27:
FYI that I find the following in the list of included libraries: `/usr/local/lib/python2.7/site-packages/numpy/...`. All of the other Python modules (and Python) are clearly using MacPorts (`/opt/local/Library/Frameworks/Python.framework/Versions/2.7/...`). You might want to remove this cruft just in case it's making a difference. I don't know if it would make a difference or not.

cn0xroot, 2017-06-28:
@michaelld thanks for the quick reply.

when i run everyone grc flow graph,python crash.i'll try to do what you said.

cn0xroot, 2017-06-28:
i try to do it again,gnuradio use `/opt/local/Library/Frameworks/Python.framework/Versions/2.7` to Execut python file 

```
 gnuradio-companion

** (process:17897): WARNING **: Trying to register gtype 'GMountMountFlags' as enum when in fact it is of type 'GFlags'

** (process:17897): WARNING **: Trying to register gtype 'GDriveStartFlags' as enum when in fact it is of type 'GFlags'

** (process:17897): WARNING **: Trying to register gtype 'GSocketMsgFlags' as enum when in fact it is of type 'GFlags'
<<< Welcome to GNU Radio Companion 3.7.11 >>>

Block paths:
	/opt/local/share/gnuradio/grc/blocks

Loading: "/Users/cn0xroot/sdr/AIR/am_decorder.grc"
Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_105900; UHD_003.010.001.001-MacPorts-Release

>>> Done

Generating: '/Users/cn0xroot/sdr/AIR/top_block.py'

Executing: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/cn0xroot/sdr/AIR/top_block.py

Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_105900; UHD_003.010.001.001-MacPorts-Release

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy soapy

>>> Done
```

and the result will be the same:python crash.

michaelld, 2017-06-28:
Can you post here or email me ( to: michael.dickens@ettus.com ) the GRC file, or point me to it if it's available online somewhere? That way I can verify myself directly, to determine whether it's a bug in GNU Radio or something local to your setup.

cn0xroot, 2017-06-28:
of course，i send email to you,and you can get it from https://github.com/cn0xroot/cn0xroot.github.io/tree/master/SDR/fm-radio

michaelld, 2017-06-28:
OK. Thx. This script works for me on my macOS 10.12 (latest everything) box using a B210 and B200mini.

Reviewing the traceback, it looks like UHD is failing during its startup search for devices.

Are you using an Ettus SDR device? If not, which SDR device are you using?

If you're using an ER SDR device, what happens when you unplug it / power it down; then plug it back in / power it up, and then run `uhd_find_devices`? Do you get a similar error?

michaelld, 2017-06-28:
Also: are you using the ports `gnuradio` / `uhd`, or `gnuradio-devel` / `uhd-devel`?

cn0xroot, 2017-06-29:
i use usrp b200 and hackrf.also using the ports gnuradio / uhd.

and i find some issues on github like this:
https://github.com/cfriedt/gnuradio-for-mac-without-macports/issues/22
https://github.com/cfriedt/gnuradio-for-mac-without-macports/issues/26

michaelld, 2017-06-29:
Can you try moving to `uhd-devel` and then `gnuradio-devel` and seeing if those fix the issue? You'll have to deactivate `uhd` and `gnuradio` first (e.g., `sudo port -f deac uhd gnuradio`).

michaelld, 2017-06-29:
Also: what does `port installed boost and active` return?

cn0xroot, 2017-06-30:
boost is active:

```
 port installed  |grep boost
  boost @1.59.0_3+no_single+no_static+python27 (active)
```

cn0xroot, 2017-06-30:
```
  gnuradio-devel @20170622_0+docs+grc+jack+portaudio+qtgui+sdl+swig+uhd+wavelet+wxgui (active)
  gqrx-devel @20170531_0 (active)
  uhd-devel @20170627_0+docs+examples+gpsd+libusb+manpages+manual+python27+test (active)
```

it work fine now. thank you so much!

michaelld, 2017-06-30:
OK. Great & I'm glad moving to the devel versions did the trick. Seems like there's an issue with the release. I'll see if I can replicate on my local setup using the releases. Thanks for being so active in helping debug this issue!

michaelld, 2017-06-30:
@mbr0wn : I can't close this issue for some reason. Could you please do so (and, maybe, tell me how to do so / what I'm missing?)? Thx!

michaelld, 2017-07-15:
@jmcorgan Please close this issue.

closed 2017-07-15 by jmcorgan

drejtiv, 2017-09-17:
I had a similar issue (9 out of 10 Python executions with my HackRF resulted in segfault.)  I followed the advice and switched to uhd-devel and gnuradio-devel which solved the problem (now maybe 1 in 20 launches produces a segfault, but the next attempt to launch will always succeed.)

michaelld, 2017-09-18:
Great! That's about what I get too for crashes. Doesn't require actual SDR hardware either. Just running GR (gnuradio-devel). I haven't tried debugging yet.