gnuradio/gnuradio #1353: uhd: unlocking flowgraph in gnuradio disconnects USRP-Sink

sugandhagupta, 2017-06-29:
@noc0lour reported: 

Running a simple flowgraph with a USRP-sink connected and performing a lock() and unlock() of the flowgraph triggers an error in the transmitter and it stop transmitting.
This error is encountered with latest uhddev/master 1b70b3e and gnuradio/master but not with uhddev/3.9-LTS fdec013 and gnuradio/master.

stdout:
```
linux; GNU C++ version 4.9.3; Boost_105600; UHD_003.011.000.git-95-g811444b1

-- X300 initialization sequence...
-- Determining maximum frame size... 1472 bytes.
-- Setup basic communication...
-- Loading values from EEPROM...
-- Setup RF frontend clocking...
-- Radio 1x clock:200
-- Detecting internal GPSDO.... No GPSDO found
-- [DMA FIFO] Running BIST for FIFO 0... pass (Throughput: 1177.5MB/s)
-- [DMA FIFO] Running BIST for FIFO 1... pass (Throughput: 1182.5MB/s)
-- [RFNoC Radio] Performing register loopback test... pass
-- [RFNoC Radio] Performing register loopback test... pass
-- [RFNoC Radio] Performing register loopback test... pass
-- [RFNoC Radio] Performing register loopback test... pass
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
locking....
unlocking...
Press Enter to quit: thread[thread-per-block[1]: <block gr uhd usrp sink (1)>]: RuntimeError: On node 0/DmaFIFO_0, input port 1 is already connected.
```

flowgraph:
```
#!/usr/bin/env python2
# -*- coding: utf-8 -*-
##################################################
# GNU Radio Python Flow Graph
# Title: Top Block
# Generated: Fri Oct  7 16:36:43 2016
##################################################

from gnuradio import analog
from gnuradio import eng_notation
from gnuradio import gr
from gnuradio import uhd
from gnuradio.eng_option import eng_option
from gnuradio.filter import firdes
from optparse import OptionParser
import time

class top_block(gr.top_block):

    def __init__(self):
        gr.top_block.__init__(self, "Top Block")

        ##################################################
        # Variables
        ##################################################
        self.samp_rate = samp_rate = 1e6

        ##################################################
        # Blocks
        ##################################################
        self.uhd_usrp_sink_0 = uhd.usrp_sink(
                ",".join(("addr=192.168.10.2", "")),
                uhd.stream_args(
                        cpu_format="fc32",
                        channels=range(1),
                ),
        )
        self.uhd_usrp_sink_0.set_subdev_spec('B:0', 0)
        self.uhd_usrp_sink_0.set_samp_rate(samp_rate)
        self.uhd_usrp_sink_0.set_center_freq(1e9, 0)
        self.uhd_usrp_sink_0.set_gain(0, 0)
        self.analog_sig_source_x_0 = analog.sig_source_c(samp_rate, analog.GR_COS_WAVE, 1000, 1, 0)

        ##################################################
        # Connections
        ##################################################
        self.connect((self.analog_sig_source_x_0, 0), (self.uhd_usrp_sink_0, 0))

    def get_samp_rate(self):
        return self.samp_rate

    def set_samp_rate(self, samp_rate):
        self.samp_rate = samp_rate
        self.uhd_usrp_sink_0.set_samp_rate(self.samp_rate)
        self.analog_sig_source_x_0.set_sampling_freq(self.samp_rate)

def main(top_block_cls=top_block, options=None):

    tb = top_block_cls()
    tb.start()
    time.sleep(3)
    print('locking....')
    tb.lock()
    time.sleep(3)
    print('unlocking...')
    tb.unlock()
    try:
        raw_input('Press Enter to quit: ')
    except EOFError:
        pass
    tb.stop()
    tb.wait()

if __name__ == '__main__':
    main()
```
@michael-west and I were able to reproduce the bug. It was traced down to a gr-uhd bug.
In the start function in the sink block, tx_streamer was not getting destroyed before getting recreated again.

noc0lour, 2017-06-30:
Are you working on the fix/have a fix already?

sugandhagupta, 2017-06-30:
I do. Will have a pull request soon.

mbr0wn, 2017-07-07:
Closing issue, deferring discussion to PR #1355.

closed 2017-07-07 by mbr0wn