gnuradio/gnuradio #1357: UHD: USRP source block does not generate stream tags when RX frequency changed via its message port.

vermillionsands, 2017-07-01:
Generally, if the USRP source block's centre frequency is altered then the output stream is tagged with the current time, frequency, and rate. However, if the centre frequency is altered via messages received on its command port then these  tags are not generated. This seems like a bug.

The root cause is that command messages use the SET_CENTER_FREQ_FROM_INTERNALS macro to set the RX frequency instead of the set_center_freq() function. The macro does not set the _tag_now flag, so no tags are generated.

dkozel, 2017-07-03:
@vermillionsands Thanks for opening this issue and for having taken a look into the root issue. The creation of a tag with the center frequency does seem like a behavior which should be consistent no matter which API is used to set the frequency.

fedjo, 2018-09-20:
+1 
I also experience the same behavior.
It would be nice to resolve this issue.  Any news about it?

Otherwise is any workaround to periodically change the RX freq of USRP source and recognize it in a way (with a tag, witha custom block) ?

seldad8, 2021-02-17:
I'm able to work around this issue by sending a "tag" command on the message port, after sending the "tune" message.

marcusmueller, 2023-12-20:
Closed per #4259

PR #4259 (2021-02-18): gr-uhd: Generate tags when frequency changed from message port.
Call set_center_freq() rather _dev->set_rx_freq() so that _tag_now is set on the change.

The end result is that the tags are generated as needed when changes are directed from the message port. set_center_freq() performs the same functionality as the original code with the addition of properly setting the _tag_now flag.

fixes #1357 
fixes #4204.
commit 4a84443c (2021-02-18): Corrected indentation

fixes #1357 and #4304
commit 61d19947 (2021-02-18): Call set_center_freq rather _dev->set_rx_freq so that _tag_now is set on the change.

fixes #1357 and #4204.
closed 2023-12-20 by marcusmueller