gnuradio/gnuradio #1361: i installed gnuradio 3.7.12 which version gr-osmocomsdr i should installed

stackprogramer, 2017-07-06:
i installed gnuradio3.7.12, i installed this version osmocomsdr
https://github.com/osmocom/gr-osmosdr/tree/2a2236cc9e97581e59cfc8018bd03d90659e6acc
but osmocom in receive it has error 

>   File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 4879, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: list contains invalid format!
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

which version should i install  gr-osmosdrthat it works with gnuradio 3.7.12 ??
thanks

stackprogramer, 2017-07-06:
in installed in newest version https://github.com/osmocom/gr-osmosdr 
gr-osmosdr  we have this error 
https://github.com/osmocom/gr-osmosdr/commit/c653754dde5e2cf682965e939cc016fbddbd45e4
root@SP:/home/sp/Downloads/gr-osmosdr-master/build# make
[  2%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o
/home/sp/Downloads/gr-osmosdr-master/lib/rtl/rtl_source_c.cc: In constructor ‘rtl_source_c::rtl_source_c(const string&)’:
/home/sp/Downloads/gr-osmosdr-master/lib/rtl/rtl_source_c.cc:224:43: error: ‘rtlsdr_set_bias_tee’ was not declared in this scope
   ret = rtlsdr_set_bias_tee(_dev, bias_tee);
                                           ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:238: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o] Error 1
CMakeFiles/Makefile2:121: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

jmcorgan, 2017-07-07:
This is a forum for reporting bugs in GNU Radio. For general support please use the discuss-gnuradio mailing list or #gnuradio on IRC/Slack.

closed 2017-07-07 by jmcorgan