gnuradio/gnuradio #1362: File Sink/Source clipping data

LaughingCabbage, 2017-07-07:
Data is clipped during a file source / file sink flow. Clipping is varies according to file sink "Unbuffered" setting. With unbuffered OFF: Input file is 37,075 bytes and output file is 36,864 bytes leaving a 211 byte loss from input to output. With unbuffered ON: Input file is 37,075 bytes and output file is 37,072 bytes leaving a 3 byte loss. Bug discovered in testing of pull [#1360](https://github.com/gnuradio/gnuradio/pull/1360). Additionally, bug was noted on email list where users could only pipe data correctly with file source set to repeat. The repeat could potentially cause a flush of the buffer holding the missing data. TEST FLOW: File Source(repeat off) -> File Sink.

mbr0wn, 2017-07-14:
I took the liberty of renaming the issue; clipping usually indicates that the samples go through but get modified.

jmcorgan, 2017-07-14:
There's a recurring, difficult to reproduce bug introduced several years ago with finite flowgraph shutdown where some parts of the graph are shutdown prematurely. Not sure if this is related but but worth considering.

LaughingCabbage, 2017-07-14:
So I face-palmed pretty hard here. The flow graph I used was set to pass float values where as my file should have been read in as a byte format. In doing so I told the file source to consume 4 bytes at a time. However, my test file's size did not divide evenly by 4 and a 3 byte remainder was never read. I've added a warning in #1381 to hopefully make this a bit more clear for users.

noc0lour, 2017-10-31:
So this is still an issue? Or is the warning enough since GNU Radio can't do anything about truncated data?

LaughingCabbage, 2017-10-31:
@noc0lour this shouldn't be an issue anymore. The warning should suffice.

frestuc, 2018-01-27:
Hi, the bug is still there. Still the sink does not dump every received byte...

noc0lour, 2018-01-27:
So this might be expected behaviour if the total size of the file is not a multiple of your item size I think?

frestuc, 2018-01-29:
No, it's truncating a lot, like 50kBytes out of 380 kBytes. Also tried to directly dump via file write but still truncating...

noc0lour, 2018-01-29:
Can you elaborate a bit? If I plug a file into my source and stream it will not read everything out of that file? Or if I stream into a file it will not write everything to file?

frestuc, 2018-01-30:
Source and sink are byte-size items, so it should be fine right?
I am trying with tx_ofdm and rx_ofdm.py connected together.

frestuc, 2018-01-30:
This is the input (small.mp4) and the output (small_out.mp4):
-rw-rw-r--  1 francesco francesco    383631 Jan 27 12:30 small.mp4
-rw-r--r--  1 francesco francesco    382560 Jan 29 19:59 small_out.mp4

frestuc, 2018-01-30:
Obviously there is not bit error rate between OFDM txer and rxer

noc0lour, 2018-01-30:
Obviously a complex flow graph like OFDM rx/tx isn't a viable test to pinpoint the issue to the file sink/source. I will hack something together and test sink/source back to back. Also the new unit tests do not indicate truncation issues. 
I feel there might be data stuck in buffering in between.

willcode, 2018-01-30:
Having an FFT in the flowgraph can drop N % FFT_SIZE samples.

frestuc, 2018-01-30:
@willcode why so?
@noc0lour what do you mean by "stuck in between"?

LaughingCabbage, 2018-01-30:
I was previously tracking this issue in #1392. The problem is not with the source/sink. This can be tested by linking the two directly. The issue lies in the packet decoder block. I was forced to move with other work, but the packet decoder would be my focus.

frestuc, 2018-01-30:
@LaughingCabbage what OFDM block do you mean in particular?

willcode, 2018-01-30:
@frestuc some blocks (e.g., FFT and packet-based ones) need a certain number of samples/bytes before they can output anything. An over-the-air receiver is constantly receiving samples, so you wouldn't have a problem with flushing, or at least not this problem. The TX side can sometimes cause problems if not all samples are flushed and the SDR stops transmitting. A better test would be to send the output of tx_ofdm.py to a file, pad the beginning and end with a bunch of zero samples, and then send the result to rx_ofdm.py.