gnuradio/gnuradio #1367: Tun/Tap block for MacOS

vamsi765, 2017-07-12:
@michaelld 
I get the following error when I have tun tap_pdu block in grc. The error persists even after installing tuntap related files from MacPort.

`
Traceback (most recent call last):
  File "gen_sink.py", line 58, in <module>
    main()
  File "gen_sink.py", line 47, in main
    tb = top_block_cls()
  File "gen_sink.py", line 30, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu('tap0', 10000, False)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/blocks/blocks_swig6.py", line 1042, in make
    return _blocks_swig6.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: tuntap_pdu not implemented on this platform
`

michaelld, 2017-07-12:
@mbr0wn : There is no "tuntap" directly on OSX like there is on Linux. There is a "tuntaposx" project that adds this capability, somehow. I've never used it, but it might be worth investigating to see if it can be used to expand the functionality of the tuntap block. Do you want me to give it a try & see what I can do for an hour or 2?

mbr0wn, 2017-07-14:
@michaelld Go for it!

jmcorgan, 2017-07-14:
@mbr0wn Same issue here; @michaelld does not show up in the list of people I can assign this to.

michaelld, 2017-07-14:
@mbr0wn OK thx I'll take a look-see.

devnulling, 2020-01-19:
@michaelld is this still an issue?

michaelld, 2020-01-19:
yes ... I've never found/made time to look into this.

michaelld, 2021-09-25:
There is no good way to do TAP/TUN on macOS that I know of. I never had time to deal with this, and will not any time in the near future. Given how old this issue is and the lack of traction on it, I'm closing it. If someone truly wants this to work, go ahead and reopen and please suggest a solution!

closed 2021-09-25 by michaelld