gnuradio/gnuradio #1368: Tag Gate error

vamsi765, 2017-07-12:
I get the following error when running the flow graph with "Tag Gate" block:
``
Traceback (most recent call last):
  File "packet_uhd_usrp.py", line 19, in <module>
    from packet_uhd_tx_rx import packet_uhd_tx_rx  # grc-generated hier_block
  File "packet_uhd_tx_rx.py", line 18, in <module>
    from packet_tx import packet_tx  # grc-generated hier_block
  File "/Users/vamsi/.grc_gnuradio/iris_packet_tx.py", line 78
    self.blocks_tag_gate_0.set_single_key(""))
                                             ^
SyntaxError: invalid syntax
``

I see this error when using the latest commit (I run on MacOS with GNURadio installed using Macports gnuradio-devel). "v3.7-MacPorts-devel-git-54e89c64(20170622)"

The error goes away when the line is commented out. [This line was not generated for people running on older version in Linux].

LaughingCabbage, 2017-07-13:
I remember this same issue from a gr-digital packet example. What flow are you using here?

vamsi765, 2017-07-14:
Any simple flow graph with the block fails.

[image]

vamsi765, 2017-07-14:
I have noticed that this fails even in Linux on my friends system, when he updated to use the latest version using PyBombs.

jmcorgan, 2017-07-14:
The fix for this was merged today.

vamsi765, 2017-07-15:
@michaelld could you update the gnuradio-devel macport?

michaelld, 2017-07-16:
Yes ... been on my queue for a bit now & I'll find time early next week; sorry for the delay!

I have to deal with some patches because the GR API was changed & the #include directories aren't ordered correctly right now in CMake -- so, the old API is picked up if it is already installed & that breaks the build. Always a good time fixing these things!

vamsi765, 2017-07-16:
Thank you @jmcorgan and @michaelld !

closed 2017-07-16 by vamsi765