gnuradio/gnuradio #1378: types.h: get rid anachronisms now that we can

marcusmueller, 2017-07-13:
Considering we can now rely on halfway recent MSVC:

we really shouldn't be doing things like

```
typedef long long               gr_int64;
```

these days. There's `cstdint` for a reason, and that is portability. Also, we<sup>TM</sup> should probably check all these types for actual usage – we can afford to break that one OOT that is strange enough to use `gr_int64` instead of `int64_t` in the 3.8 changeover, if we can avoid lunging around completely useless/unused `typedef`s.

This is even worse in `pmt`, where we really need the unambiguity of types like `uint64_t` instead of `unsigned long long` (by the way, the fact that `long long` is standardized is a C++11 extension, so "backwards compat" doesn't 100% fly).

jmcorgan, 2017-07-13:
I'm all for cstdint cleanliness, even in 3.7 if possible.

mbr0wn, 2017-07-14:
`cstdint` requires C++11, but `int32_t` and friends do not. We can include `stdint.h`, which is compatible.

jmcorgan, 2017-07-14:
@marcusmueller Will you take this on?

marcusmueller, 2017-07-14:
@jmcorgan yes. Please assign.

jmcorgan, 2017-07-14:
Great, thanks.

jmcorgan, 2017-07-14:
@mbr0wn For some reason I continue to have a problem with assigning from the menu drop down; I get a list of 10 people but @marcusmueller is not on it.

marcusmueller, 2017-07-14:
This change leaves gnuradio/types.h a hollow shell, its pale shadow of utility being #include <stdint.h> ; should I clean up inclusion of types.h in the tree and replace with stdint.h where actually used for that? It's included without reason in several places so I'm going to remove these #includes, anyway.

marcusmueller, 2017-07-14:
Files that `#include <gnuradio/types.h>`, but don't actually mention any `[u]int{width}_t`: 

```
%> ag -L 'int[[:digit:]]*_t' $(ag -l "<gnuradio/types")
gr-filter/lib/hilbert_fc_impl.h
gr-filter/lib/qa_fir_filter_with_buffer.cc
gr-filter/include/gnuradio/filter/hilbert_fc.h
gr-filter/include/gnuradio/filter/pm_remez.h
gnuradio-runtime/lib/misc.h
gnuradio-runtime/include/gnuradio/runtime_types.h
gnuradio-runtime/include/gnuradio/expj.h
gnuradio-runtime/include/gnuradio/message.h
gnuradio-runtime/include/gnuradio/misc.h
gr-fft/include/gnuradio/fft/goertzel.h
gr-digital/include/gnuradio/digital/crc32.h
gr-channels/include/gnuradio/channels/channel_model2.h
gr-channels/include/gnuradio/channels/cfo_model.h
gr-channels/include/gnuradio/channels/sro_model.h
gr-channels/include/gnuradio/channels/channel_model.h
gr-channels/include/gnuradio/channels/fading_model.h
gr-channels/include/gnuradio/channels/selective_fading_model.h
gr-channels/include/gnuradio/channels/dynamic_channel_model.h
```

marcusmueller, 2017-07-14:
https://github.com/gnuradio/gnuradio/pull/1380

mbr0wn, 2017-07-14:
@jmcorgan Not sure why you can't assign to Marcus. We both have the same level of privilege. :confused:

mbr0wn, 2017-07-14:
@marcusmueller We still use `gr_vector_void_star` and friends though.

mbr0wn, 2017-07-14:
Maybe we should run https://include-what-you-use.org/ over our codebase....

jmcorgan, 2017-07-14:
Oh now that's interesting, it could be something we run alongside coverity, etc. in the CI system.

closed 2021-03-21 by marcusmueller