gnuradio/gnuradio #1383: logger.h change causes heap corruption in certain instances

gnieboer, 2017-07-14:
Commit 08ab03d781135481f15afc1ef0754c60d4ccad42 introduced a heap corruption issue for OOT modules in certain build configurations and certain compilers.

logger.h lines 59-61 cause heap corruption on systems where sizeof(std::string) != sizeof(void *) and logging is not requested by a module.

Compilers affected appear to include GCC 5.0 and MSVC, or in general any compiler that uses short string storage on the stack.

The change in this commit impacts scenarios where GNURadio was built with logging support (but not log4cpp) but the module did not explicitly set ENABLE_GR_LOG flag.  Prior to the commit, it appears ENABLE_GR_LOG would be defined, essentially inheriting from the GNURadio configuration.  Post commit this flag is undefined.

This then results in an issue later in logger.h where logger_ptr is defined as a void* instead of a std::string for the OOT module, a mismatch from the GNURadio library

This then results in a  different definition of gr::block, specifically a difference in sizeof(block), so calls to malloc() and free() from different libraries result in different memory allocations and end up in heap corruption.

A workaround is to explicitly define ENABLE_GR_LOG for all OOT modules.
Two possible fixes I see:
1- Refactor logger.h to keep ENABLE_GR_LOG defined if GNURadio was built with it
2- Refactor logger_ptr and usage thereof to ensure that it always represents something the size of a system pointer so a public api isn't affected by a build flag.

jmcorgan, 2017-07-14:
Without thinking deeply about it I'd prefer #1. Also, all this conditional code has been removed already and is always "on", with log4cpp now a dependency, on the next branch.

michaelld, 2017-07-15:
Great sleuthing work! Isn't the actual issue the use of std::string, since the other 2 variants use a pointer -- which will be of the same size no matter which is used? Seems like replacing the std::string version with std::string* or something like that would be the natural way to go to guarantee sizes are the same. This would, of course, be for just maint and master, not next -- since, as JC writes this issue has already been resolved on next.

gnieboer, 2017-07-15:
I agree with #1 as well for 3.7.x, and leave fix #2 for 3.8.

jmcorgan, 2017-07-15:
In 3.8 there is no conditional code--logging is always enabled and log4cpp is a required dependency. Doesn't the issue go away then?

gnieboer, 2017-07-15:
Yes, it sounds like what's already planned implements fix #2 in 3.8.

michaelld, 2017-07-16:
I'd say to do #2 right now for 3.7, since it should be almost trivial. I'll test this out when I do updates Monday / Tuesday ... I don't think the logger_ptr is even used when logging is enabled but not using log4cpp. But I haven't looked through -all- of the code to verify. Simple test. More when I know it.

Flamewires, 2017-07-17:
For what its worth I've gotten this using gcc 6.3 as well. Thanks for digging into debugging it Geof.

michaelld, 2017-07-23:
I've been working on replicating this issue on OSX, with no success. I can verify that when logger.h.in line `typedef std::string logger_ptr` is changed to `typedef void* logger_ptr`, GR builds without issue. I would also guess that any OOT will work correctly then because all 3 typedefs of logger_ptr are pointers. The type used for this change doesn't matter because it's not used, so why not make it the same as for when logging is turned off entirely? Can others try this change to see if it works on this OS(s)? If we can verify this fix, then I'll do a PR for it quickly.

gnieboer, 2017-07-25:
Win64 build scripts completed without a problem with my previous workaround removed and the line above changed as you described, a quick check indicates gqrx worked as expected, but I be a bit more thorough tonight.

gnieboer, 2017-07-26:
Did more thorough testing, no issues discovered.

michaelld, 2017-07-26:
Excellent. Thanks for testing! I'll create a PR for this change, along with my others related to logging / log4cpp.

michaelld, 2017-07-28:
This fix and much more: https://github.com/gnuradio/gnuradio/pull/1398

bastibl, 2017-07-28:
I don't get the problem with 4.) IIRC, the absence of this include caused many build problems in the past and would require adding the include to every submodule since every block is indirectly including log4cpp:
https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/include/gnuradio/logger.h.in#L171

michaelld, 2017-07-29:
If you check, the submodules that use logging already have LOG4CPP_INCLUDES included. I didn't put them there. Please test and verify. I've done so many times with the various logging options enabled / disabled to verify functionality ... all now work for me with these changes, admittedly on OSX but CMake is generic enough they should work on the other supported platforms too.

michaelld, 2017-07-29:
I think before https://github.com/gnuradio/gnuradio/commit/08ab03d781135481f15afc1ef0754c60d4ccad42 we had to have the LIBRARIES and INCLUDES included pretty much everywhere. This commit allows better selective use of logging, so that we don't need them everywhere any longer. This PR fixes issues with that commit & also related logging issues mostly with the CMake parts (minimal compile-time issues fixed).

bastibl, 2017-07-29:
I don't know what you mean with selective use of logging, but I had the impression that every block includes `block.h`, which includes `logger.h`, which references log4cpp if logging is enabled. Therefore, I don't understand the comment about submodules that use logging. I though all submodules need the log4cpp includes. IIRC, that was the problem in several issues.
I added the include to `GNURADIO_RUNTIME_INCLUDE_DIRS` because I belief it should contain all paths that are required by its public includes (in the `include` dir). Otherwise dependecy tracking in submodules might become a mess.

michaelld, 2017-07-29:
Drat. This change works because the LOG4CPP_INCLUDE_DIRS is the same as BOOST_INCLUDE_DIRS for my specific install. If they were different then the submodules that do not add LOG4CPP_INCLUDE_DIRS would fail to build. So back to the drawing board, at least partially.

That said, we shouldn't add LOG4CPP_INCLUDE_DIRS to GNURADIO_RUNTIME_INCLUDE_DIRS because we can't guarantee that GNURADIO_RUNTIME_INCLUDE_DIRS will be used between local-to-build includes and external includes. Currently in some submodules it is, and in some it's not. LOG4CPP_INCLUDE_DIRS is, by definition, external to the GR build & hence any *_INCLUDE_DIRS after it must be for external directories only.

There needs to be a better way to split such variables to make this setup work!

"Selective use of logging" is for OOT modules, but can be applied inside GR submodules too if desired. For example if the OOT / submodule code does not define ENABLE_GR_LOG, then logging will be disabled in it, no matter if it was enabled in GR-runtime. Similarly, if ENABLE_GR_LOG was not defined for GR-runtime, then no OOT can use logging no matter the setting of ENABLE_GR_LOG.

bastibl, 2017-07-29:
> That said, we shouldn't add LOG4CPP_INCLUDE_DIRS to GNURADIO_RUNTIME_INCLUDE_DIRS because we can't guarantee that GNURADIO_RUNTIME_INCLUDE_DIRS will be used between local-to-build includes and external includes.

True, but I thought it's a step in the right direction. We might want to add `LOG4CPP_INCLUDE_DIRS` to `GNURADIO_RUNTIME_INCLUDE_DIRS` and record the variable in ` GnuradioConfig.cmake`. Then an OOT gets the right directories when doing

```
set(GR_REQUIRED_COMPONENTS RUNTIME)
find_package(Gnuradio)
```

In general, I think, `GNURADIO_X_INCLUDE_DIRS` should contain the directories that are needed by the headers in `gr-x/include/gnuradio/x/`. These variables should be recorded during build and be accessible to OOTs when enabling submodule X.

> Currently in some submodules it is, and in some it's not. LOG4CPP_INCLUDE_DIRS is, by definition, external to the GR build & hence any *_INCLUDE_DIRS after it must be for external directories only.

In next that is already ["cleaned up"](https://github.com/gnuradio/gnuradio/pull/1195/commits/51e565532f1f867baacc004222971fcf3703773b), but, AFAIS, that's not the correct approach.

michaelld, 2017-07-29:
So for OOTs I agree that one should set the GNURADIO_X_INCLUDE_DIRS as you say.

The issue here is internal to the GR repo. If each GNURADIO_X_INCLUDE_DIRS were set and some contained external dirs, then there is no way to order them such that all external dirs come after internal dirs. This ordering makes a difference on systems where an already-existing GR install is in place while the new one is being built. For example, I haven't updates GR in MacPorts since the last few commits because the API changed & right now the include dirs ordering picks up the old dirs with the old API instead of the new dirs with the updated API, and so the build fails.

Yes, next branch is already forces logging and log4cpp, IIRC. I don't know that that's the correct approach either, but it's where GR is moving.

bastibl, 2017-07-30:
I see, so if one in-tree submodule depends on two submodules it could lead to problems (otherwise it could be solved by reordering includes). But even with that problem, the definition of `GNURADIO_X_INCLUDE_DIRS` might still be better than, for example, conditionally adding `LOG4CPP_INCLUDE_DIRS` to every submodule. 

Since cmake doesn't seem to support it, how about replacing `include_directories` with a macro that iterates over the directories and does `include_directories(BEFORE ...)` for every subdirectory of `PROJECT_SOURCE_DIR` and `PROJECT_BINARY_DIR`, and `include_directories(AFTER ...)` for the others?

michaelld, 2017-07-30:
Hmnm .. that's a good idea. It's worth a try. I'll look into it next week.

michaelld, 2017-08-22:
Long time! @bastibl Yes, we can do this last suggestion & it seems to work really well! I have some testing to before pushing a PR, but overloading `include_directories` is definitely a good way to fix the include ordering for real & once and for all.

michaelld, 2017-08-22:
Given the complexity of this change & the testing required, I'm going to create a new PR for the logger_ptr issue to get it fixed, separate from the fix in #1398 -- and, then that PR will just be this overloaded macro & related changes.

jmcorgan, 2017-08-27:
So, with #1438 merged, what's your next step?

michaelld, 2017-08-27:
I need to get the overloaded macro pushed, which I think is best in a separate PR for folks to do testing on. Once that's in place, then I'll revisit this issue. I think that ordering makes the most sense. I should get the overloaded macro PR out tomorrow.

michaelld, 2017-08-27:
See #1447 for my PR.

michaelld, 2020-03-27:
I'm guessing since my PR was merged that this issue is resolved. If it or something related to it is still an issue, then let's open a new issue ... this one is ancient!

closed 2020-03-27 by michaelld