gnuradio/gnuradio #1388: Make failing for new gnuRadio install on ubuntu for interpolating_resampler.cc

Sandeep-Sarma, 2017-07-18:
On checking out the latest version (commit: bd611224ba4c2356c1ad4c3359781682c5c121da), I get the following errors on make:

make[2]: *** [gr-digital/lib/CMakeFiles/gnuradio-digital.dir/interpolating_resampler.cc.o] Error 1
make[1]: *** [gr-digital/lib/CMakeFiles/gnuradio-digital.dir/all] Error 2

Error:
--------
gnuradio/gr-digital/lib/interpolating_resampler.cc:392:70:   instantiated from here
/usr/include/c++/4.6/bits/stl_vector.h:1080:4: error: invalid conversion from ‘int’ to ‘std::vector<gr::filter::kernel::fir_filter_fff*>::value_type {aka gr::filter::kernel::fir_filter_fff*}’ [-fpermissive]

usr/include/c++/4.6/bits/stl_vector.h:1122:7: error:   initializing argument 2 of ‘void std::vector<_Tp, _Alloc>::_M_fill_initialize(std::vector<_Tp, _Alloc>::size_type, const value_type&) [with _Tp = gr::filter::kernel::fir_filter_fff*, _Alloc = std::allocator<gr::filter::kernel::fir_filter_fff*>, std::vector<_Tp, _Alloc>::size_type = unsigned int, std::vector<_Tp, _Alloc>::value_type = gr::filter::kernel::fir_filter_fff*]’ [-fpermissive]

jmcorgan, 2017-07-18:
@awalls-cx18

jmcorgan, 2017-07-18:
What is your OS version?

awalls-cx18, 2017-07-18:
On July 18, 2017 10:35:45 AM EDT, Johnathan Corgan <notifications@github.com> wrote:
>@awalls-cx18 

This compiles on Fedora 26 and for the git  PR buildbot without error.

The problem appears to be the version of C++ in use is being very particular about wanting a real pointer value and not NULL, which is just an integer.

The simple solution is to remove the 2nd argument (NULL) in the 8 vector initializer calls in that file.

-Andy

Sandeep-Sarma, 2017-07-18:
Thanks. Let me try that. My OS version is Ubuntu 12.04.

jmcorgan, 2017-07-18:
I'm going to be pushing a PR for fixing this that should work for you, but please remember that Ubuntu 12.04 is no longer supported by us or Canonical.

Sandeep-Sarma, 2017-07-18:
Thanks Sir. Point noted.

jmcorgan, 2017-07-18:
@Sandeep-Sarma PR #1389 has the recommended changes; please confirm this solves your problem as we do not test gcc 4.6 or Ubuntu 12.04.

PR #1389 (2017-07-20): gr-digital: fixed casting failure compile error with gcc 4.6
Fixes #1388.
closed 2017-07-20 by jmcorgan