gnuradio/gnuradio #1390: gnuradio-3.7.10.2 & gnuradio-3.7.11 doesn't support compiling gqrx-2.6.1 by cause of log4cpp

hkopp64, 2017-07-19:
Hello,

After succesfull compile en install of versions gnuradio-3.7.10.2 and gnuradio-3.7.1, I get underlaying compile error in gqrx-2.6.1.

Does someone earlier dealing with this issue and know about to solve it, because in gnuradio-3.7.10.1 I had no compile errors in gqrx-2.6.1

[ 89%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/receiver_base.cpp.o
[ 90%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/wfmrx.cpp.o
[ 92%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_device_list.cc.o
/usr/local/src/gqrx/src/pulseaudio/pa_device_list.cc: In member function ‘int pa_device_list::populate_device_list()’:
/usr/local/src/gqrx/src/pulseaudio/pa_device_list.cc:129:49: warning: ‘pa_op’ may be used uninitialized in this function [-Wmaybe-uninitialized]
if (pa_operation_get_state(pa_op) == PA_OPERATION_DONE) {
^
[ 93%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_sink.cc.o
[ 94%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_source.cc.o
[ 96%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_icons.cpp.o
[ 97%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_textfiles.cpp.o
[ 98%] Building CXX object src/CMakeFiles/gqrx.dir/gqrx_automoc.cpp.o
[100%] Linking CXX executable gqrx
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/gqrx.dir/applications/gqrx/main.cpp.o: undefined reference to symbol '_ZN7log4cpp8Appender29AppenderMapStorageInitializerD1Ev'
/usr/lib64/liblog4cpp.so.5: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/CMakeFiles/gqrx.dir/build.make:1634: recipe for target 'src/gqrx' failed
make[2]: *** [src/gqrx] Error 1
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/gqrx.dir/all' failed
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

I used OS opensuse leap 42.3 64bit KDE4 and have no compile errors in gnuradio-3.7.10 t/m 3.7.11 and all gnuradio blocks where succesfully installed in both versions.

Thanks in advance

Huib

jmcorgan, 2017-07-19:
We're hoping to get this fixed for a 3.7.11.1 maintenance release.

mbr0wn, 2018-12-20:
Pretty sure this was fixed.

closed 2018-12-20 by mbr0wn