gnuradio/gnuradio #1392: gr-digital: packet.py withheld packet

LaughingCabbage, 2017-07-20:
I'm currently tracking a bug that retains a single packet within the hierarchy of the packet_encoder/packet_decoder system. Given a small input file of the default packet size (512 bytes), the flow graph will not pass a single packet and requires a second packet to push the first out. Additionally, the encoder requires additional padding to be added to satisfy the default payload length for consumption if the transmission is less than the default payload length (<512 bytes). My initial theory is that the encoder/decoder has a chain of blocking reads (msgq delete) from two message queues within the same thread. I'm still tracing the flow of data but I will post updates as I progress.

luzpaz, 2023-07-23:
Can this ticket be closed ?

marcusmueller, 2023-12-20:
removed from 3.8

closed 2023-12-20 by marcusmueller