gnuradio/gnuradio #1395: GNU Radio should complain when it detects "unknown" settings in its conf. files

nakic, 2017-07-27:
I am getting started working on OOT blocks and have added the following to ~/.gnuradio/config.conf in order to be able to see my blocks in action:

    local_block_path=/usr/local/share/gnuradio/grc/blocks

Lost the better part of the day looking for a reason why my blocks wouldn't appear in the GRC block list ("wrong file? wrong dir? bad module XML? GRC bug?...") until someone on stackoverflow pointed out that it was "blocks", not "block", plural: "local_blocks_path".

It seems to me it might be easy to spare future OOT block developers the frustration and lost time by implementing (extending?) conf. file validation. In general, I would consider the following reasonable:

* check that each setting is one GNU Radio recognises (this would have saved me)
* check that the values are sensible (directories exists, values are in range etc.)

GRC would output some kind of warning somewhere (in the CLI, in the GUI, whatever) when it stumbles into a problem like this. Maybe it already does this, but if it does, it was to subtle for me to notice the warning.

marcusmueller, 2017-07-28:
That's impossible by design.

There's no "listing" of "correct" keys – the program, whenever it needs a setting from the configuration file, just asks for it. The parser has no way of knowing if, anywhen in the future, a specific configuration key might be needed. Hence, there's no a-priori "wrong" keys.

It goes without saying that it'd be nice to have a well-documented config file. You'll find at least a reasonably self-explaining config file in /etc/gnuradio/conf.d/grc.conf.

skoslowski, 2017-07-28:
Also, GRC prints out the block paths in use.

nakic, 2017-07-29:
@marcusmueller I'm not familiar enough with how GNU Radio works to be able to weigh in, but fair enough, just thought I'd suggest it if it could save someone some time and trouble down the line.

@skoslowski Can't check right now, but haven't noticed it if it does. Even so, that would just address this specific variable, not solve the problem in general, but as Marcus pointed out, that can't be done.

I guess the issue can be closed, then.

mbr0wn, 2017-08-02:
Even if we had a way of curating a white list of options (which would be really hard to maintain), we explicitly don't make any guarantees on what's in the config file. This gives developers the option to add config keys as they need for their purposes.
`local_blocks_path` is not a required key, either. GRC has other ways of finding blocks, so it's a valid option to simply not set it.

As you say, let's close this.

closed 2017-08-02 by mbr0wn