gnuradio/gnuradio #1408: GPS tracking on a map over mobile app software using RF

MisterOizo, 2017-08-04:
Hello, I tried to search for any mobile applications that are done or in progress that used this or any other software to transfer mobile GPS data over radio.

For example, there is a map on the mobile app, where you can see each mobile devices' location as it sends it over to the other devices over the radio frequencies.

Are there any feature like this in GNU Radio or any similar projects? I'm asking because in crowded events the normal mobile data frequencies are so heavily used that they are unavailable, so other radio frequencies might be less congested. Thank you.

jmcorgan, 2017-08-04:
The issue tracker here is for reporting bugs in GNU Radio. For support or general questions like this, please use our mailing list, IRC channel, Slack, or StackOverflow. Thanks.

closed 2017-08-04 by jmcorgan

MisterOizo, 2017-08-04:
OK, sorry, as I thought it'd could considered as a feature request if it's not existing. I'll post it on the discuss-gnuradio then.