gnuradio/gnuradio #1411: Vector Sink: data() isn't thread-safe

marcusmueller, 2017-08-05:
The vector sink is commonly used to get data out of a flow graph. Internally, it's just a `std::vector` that keeps getting filled with items.

Now, data() *can* be called during runtime, and it's common practice to do so.

But: that leads to race conditions, since the `work` thread might be filling the vector while it's being read. Worse, even, since `std::vector` will allocate larger buffers when things grow, and move over memory, usage of `data()` might even lead to a race condition that implements a read after free... uh oh.

Simple solution: *Use Mutexes*.

jmcorgan, 2017-08-08:
> and it's common practice to do so.

I'm not sure how common this is; vector sinks are mostly found in QA code with data() being called after the flowgraph is over. Protecting it with a mutex though is proper. I'd check if there are any other public member functions that need to pick up the mutex at the same time.

marcusmueller, 2017-08-08:
You mean "other public functions" of the vector_sink, or globally in GNU
Radio? Because for the latter: yes, there are, and I've been avoiding to
fix those, because I'd rather bring forward a
message-handler-instead-of-calling-methods architecture than going in
and fixing every block out there.

jmcorgan, 2017-08-08:
I was referring to vector_sink only. The global issue is GNU Radio will be dealt with a different way as you've noted.

PR #1445 (2017-08-27): added a mutex to the data extraction, clear and work methods
fixes #1411
closed 2017-08-27 by marcusmueller