gnuradio/gnuradio #1413: wxgui in next branch

matix2120, 2017-08-06:
Hello,
what happend to wxgui in next branch? Are you planning to add this? I've just managed to compile gnuradio on Mac OS Sierra after fighting with it for the whole weekend, and it looks like it's impossible to build it using master. So I have to stick with next branch for now.

jmcorgan, 2017-08-07:
The gr-wxgui components have been deprecated for a very long time and have (as you've seen) already been removed on the next branch for the 3.8 release.

I'm not familiar with MacOS so I couldn't say what your problem with the master branch might be. @michaelld is our maintainer there.

matix2120, 2017-08-07:
Ok, I've already found that in the meantime. I just thought that all the wx blocks were deleted without porting them to qt, but now I see that some names just changed.

Regarding building on MacOS I'll create another issue with relevant name. This one can be closed.

closed 2017-08-07 by jmcorgan