gnuradio/gnuradio #1416: Python crash "no current thread" on brach next

matix2120, 2017-08-07:
Hello
I've build Gnuradio from sources (branch next) on MacOS 10.12.5 and I'm facing some issues - mainly crashing python when trying to use some QT block causing whole gnuradio-companion close.
I'm using homebrew and installed all the dependencies including Python. 
These are paths to all Python versions (don't know whether gnuradio uses python2 or python2.7 binary):
> $ type -a python
> python is /usr/bin/python
> $ type -a python2
> python2 is /usr/local/bin/python2
> $ type -a python2.7
> python2.7 is /usr/local/bin/python2.7
> python2.7 is /usr/bin/python2.7

After running:
`CC=/usr/bin/llvm-gcc CXX=/usr/bin/llvm-g++ cmake -DPYTHON_EXECUTABLE=/usr/local/bin/python2.7 -DPYTHON_INCLUDE_DIR=/usr/local/Frameworks/Python.framework/Versions/2.7/Headers -DPYTHON_LIBRARY=/usr/local/Frameworks/Python.framework/Versions/2.7/Python -DSPHINX_EXECUTABLE=/usr/local/bin/rst2html-2.7.py  DGR_PYTHON_DIR=/usr/local/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages ..`

All the modules are enabled despite gr-comedi, but as fair as I know this is not a problem.

`make test` shows only one failed test: 203 - qa_zeromq_pubsub (Failed)

`$ ctest -V -R qa_zeromq_pubsub`

> 203: Test command: /bin/sh "/usr/local/gnuradio/build/gr-zeromq/python/zeromq/qa_zeromq_pubsub_test.sh"
> 203: Test timeout computed to be: 9.99988e+06
> 203: F
> 203: ======================================================================
> 203: FAIL: test_001 (__main__.qa_zeromq_pubsub)
> 203: ----------------------------------------------------------------------
> 203: Traceback (most recent call last):
> 203:   File "/usr/local/gnuradio/gr-zeromq/python/zeromq/qa_zeromq_pubsub.py", line 55, in test_001
> 203:     self.assertFloatTuplesAlmostEqual(sink.data(), src_data)
> 203:   File "/usr/local/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 88, in assertFloatTuplesAlmostEqual
> 203:     self.assertEqual (len(a), len(b))
> 203: AssertionError: 0 != 1000
> 203:
> 203: ----------------------------------------------------------------------
> 203: Ran 1 test in 0.517s
> 203:
> 203: FAILED (failures=1)
> 1/1 Test #203: qa_zeromq_pubsub .................***Failed    1.01 sec

Don't know if it is really a problem?

`sudo make install` shows no errors.

Just after gnuradio-companion start I get errors:

> Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
> Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
> Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
> Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
> Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
> Warning: docstring loader crashed too often

And that's the first Python crash reported by OS, but gnuradio-companion is still working. But when I try to use some QT block, for example _QT GUI Frequency Sink_, gnuradio-companion immediately closes and Python crash with following error in the console:

> Fatal Python error: PyThreadState_Get: no current thread
> [1]    29514 abort      gnuradio-companion

Here is most interesting part of Python crash report:

> Application Specific Information:
> abort() called
> 
> Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
> 0   libsystem_kernel.dylib        	0x00007fffa45c4d42 __pthread_kill + 10
> 1   libsystem_pthread.dylib       	0x00007fffa46b2457 pthread_kill + 90
> 2   libsystem_c.dylib             	0x00007fffa452a420 abort + 129
> 3   org.python.python             	0x0000000121ca288e Py_FatalError + 49
> 4   org.python.python             	0x0000000121ca1d88 PyThreadState_Get + 28
> 5   org.python.python             	0x0000000121c9ec82 Py_InitModule4_64 + 62
> 6   _qtgui_swig.so                	0x00000001216dc48f init_qtgui_swig + 735
> 7   org.python.python             	0x00000001084b2e0c _PyImport_LoadDynamicModule + 150
> 8   org.python.python             	0x00000001084b1ba6 import_submodule + 273

I can include full report if you want.

What is more I'm able to use other blocks, like for example WBFM Receiver, and there is no error while adding it.

Can you help me fixing this?

bastibl, 2017-08-08:
That usually means that something uses or is linked against the system installation of Python. Your cmake command looks good. Some things you could check:
- make sure that there is only one GNU Radio installation, i.e., you don't have any old installations
- your command worked and `otool -L libgnuradio-qtgui<version>.dylib` lists only the Python version of Homebrew.
- if you compiled any GNU Radio dependencies yourself, make sure that they are also linked against Homebrew's Python
- make sure GNU Radio uses the correct Python executable (open gnuradio-companion, it's just a Python script, and check the shebang)
- never had this, but maybe you use Python modules that are compiled against system Python

matix2120, 2017-08-11:
Ok, so I've started from scratch, and reinstalled almost all again and it worked. I'm pretty sure that in previous attempts I did everything the same, but as we can see I must have forgotten something.

Unfortunately I'm only partially glad, because when I try to use rtl-sdr as a source whole gnuradio randomly crashes when I'm executing the flow graph. I've noticed more crashes when I'm modifying RF gain from default 10 dB to something else, but sometimes it crashes also with 10 dB. I have to re-run the flow graph a few times to make it work. Sometimes I have to even restart gnuradio or disconnect and connect rtl-sdr. Did you ever noticed such a thing?

I'm using master from git://sdr.osmocom.org/trac/wiki/rtl-sdr but I had to make a little patch, because linker couldn't find function related to antenna bias. So I commented it out since my rtl-sdr doesn't support bias voltage.

closed 2018-12-13 by mbr0wn