gnuradio/gnuradio #1420: Incorrect Asio buffer use in tcp_connection

v1bri, 2017-08-10:
I was strolling through some Asio code when I came across a small bug in `tcp_connection`. The `boost::asio::async_write()` call requires that the underlying buffer remain valid until the write handler has been called. However the current version uses a local `std::vector` that will go out of scope once the calling method returns.
https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/tcp_connection.cc#L58

A fix could involve allocating a buffer in `tcp_connection::send()` and releasing it in `tcp_connection::handle_write()`. I've opened #1419 with that change and a unit test.

mbr0wn, 2017-10-03:
#1419 was merged, can this issue be closed?

marcusmueller, 2023-12-20:
After another rewrite of TCP blocks: closing

closed 2023-12-20 by marcusmueller