gnuradio/gnuradio #1430: REGEX, mode REPLACE needs at least 6 arguments total

drom, 2017-08-17:
Linux, OpenSuse Tumbleweed. Kernel 4.11.8-2

Created simple OOT module and simple block in it.

Trying to do cmake and getting CMake error:

```sh
CMake Error at cmake/Modules/GrTest.cmake:48 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
```

Could you suggest what goes wrong?

Full log:
```sh
drom@drom:~/work/github/drom/OFDM/gr-kiss_ofdm/build> cmake ..
-- The CXX compiler identification is GNU 7.1.1
-- The C compiler identification is GNU 7.1.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.64.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib64/libcppunit.so;dl  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") 
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
 * INCLUDES=/usr/include
 * LIBS=/usr/lib64/libgnuradio-runtime.so;/usr/lib64/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /usr/lib64/libgnuradio-runtime.so;/usr/lib64/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
-- Found Git: /usr/bin/git  
CMake Error at cmake/Modules/GrTest.cmake:48 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  lib/CMakeLists.txt:76 (GR_ADD_TEST)

-- Using install prefix: /usr/local
-- Building for version: v1.0-compat-xxx-xunknown / 1.0.0git
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig  
-- Found PythonLibs: /usr/lib64/libpython2.7.so (found suitable version "2.7.13", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.13", minimum required is "2") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of size_t
-- Check size of size_t - done
-- Check size of unsigned int
-- Check size of unsigned int - done
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
-- Configuring incomplete, errors occurred!
See also "/home/drom/work/github/drom/OFDM/gr-kiss_ofdm/build/CMakeFiles/CMakeOutput.log".
```

BrashEndeavours, 2017-08-18:
Error is in this line:
`string(REGEX REPLACE "\\$\\(.*\\)" "${CMAKE_BUILD_TYPE}" path "${path}")`

Since REGEX needs certain args, try printing the `path` and `CMAKE_BUILD_TYPE` variables and make sure those are set properly?

drom, 2017-08-18:
I am following exactly the tutorial steps from here: https://wiki.gnuradio.org/index.php/Guided_Tutorial_GNU_Radio_in_C%2B%2B
The `cmake/Modules/GrTest.cmake` is automatically generated by GNU Radio boilerplate generator I guess.
I tried the same steps on two different Linux systems wit exactly the same environment.

drom, 2017-09-13:
Looks like `CMAKE_BUILD_TYPE` is not set.

Uncommenting line `CMakeLists.txt:35` solves an issue.

Was:
`#IGNORE    set(CMAKE_BUILD_TYPE "Release")`

Now:
`set(CMAKE_BUILD_TYPE "Release")`

mbr0wn, 2017-10-03:
Which CMake version do you have? And did the uncomment fix this issue for you?

devnulling, 2020-01-21:
Closing this as it is most likely no longer an issue as the cmake file has been updated several times to more modern methods. If this is still an issue feel free to reopen it or open a new issue.

closed 2020-01-21 by devnulling