gnuradio/gnuradio #1431: gr-blocks: tags_strobe nsamps callback is broken

mrjacobagilbert, 2017-08-18:
There is an issue with the tags_strobe block which is caused by incorrect tracking of data indexes. Still looking into it, but it is easy to cause incorrect behavior when changing this value. This is particularly prevalent at low sample rates. There is a flow graph [here](https://github.com/mrjacobagilbert/gr-misc_debug_utils/blob/master/examples/tags_strobe_issue.grc) that demonstrates this behavior (run and modify the 'tag_period' by clicking the up arrow a dozen times or so).

777arc, 2017-08-19:
Yeah it looks like d_offset = 0 should have never been in the setter.  I moved it to the constructor and made a PR.

Now at first this change did not seem to work for me, but then I realized it was because tags_strobe was creating tens of thousands of samples and tagging them (because the output buffer is tens of thousands large), and then the throttle is slowly pushing them through the flowgraph.  So if you change nsamps and have a low sample rate, then it can take forever to see your change be applied, because it's still feeding tagged samples that were created before you changed nsamps.  So the solution there is to set the max output buffer on tags_strobe to something on the order of your sample rate or lower, so like 100 in your example.  That way you don't have to wait more than 1 second or so before your change works its way through.  

Feel free to sanity check that PR if you have time, but it seemed to fix it.

mrjacobagilbert, 2017-08-19:
Looks like it functions correctly now. The 'broken' behavior was the duplicate tags, not the delay so that looks like a good fix to me.

mrjacobagilbert, 2020-08-31:
This was fixed by #1433.

PR #1433 (2017-08-27): Fix tags_strobe nsamps setter bug
Fixes #1431 - nsamps setter bug that causes a bunch of extra tags to be added, because it was setting d_offset = 0 (d_offset tracks where it currently is in the input buffer).  Setting d_offset to zero should happen in the constructor, not the nsamps setter.
closed 2020-08-31 by mrjacobagilbert