gnuradio/gnuradio #1442: filters and other stateful blocks need to be able to react to some `reset` tag

marcusmueller, 2017-08-27:
From a discussion with @awalls-cx18 on IRC:

* FFT filters and such are nice, but unusable for bursty operation, since there's no way to clear internal state
* this leads to own implementations and thus, code duplication

Solution I propose:

* Add a GR-wide standard `reset` tag
* On that tag, the scheduler nulls out the input buffer history of a block before calling it. This might require only calling work with items up to the tag position, then modifying the buffer, then calling work again.
* Blocks with internal state SHALL reset that state on the sample they encounter it.
* We *document* that in a GRIPE.

mrjacobagilbert, 2017-08-27:
Followup on IRC: This will have some impact on performance as all blocks adhering to this will now need to check for and act on tags. The actual impact of this is not clear to me but should be noted.

BrashEndeavours, 2017-08-27:
The performance offset could be mitigated for non reset-checking blocks by using a *reset enable* flag in the constructor/grc dialog.

marcusmueller, 2017-08-27:
Hm, that sounds much more like a flag that should have much more
universal application.

I'm thinking of something like a tag filter that you can set to
different values:

* ALL (which would be current behaviour)
* NONE (which means that this block will never call get_tags_in_range;
thus, input tags could directly be handed over to the next block(s)'s
tag container, according to the tag_propagation_policy_t that we already
have)
* LIST (+ a list of tags that your block is interested in. Tags not in
this list would be handled like in the NONE case)

This way, zero tag handling overhead for non-tag-interested blocks, only
tags that are of relevance to a specific block would be copied into the
block's tag map, and blocks that just need all tags for debug, transport
or storage reasons (tagged file sink, ZMQ sinks, tag debug) would still
work.

nowls, 2017-09-15:
For the FFT filter case, I believe a first requirement is writing a flush/clear/reset method. Second, there's a question of implementation. In the case of the FFT filter, it contains its own history/state whereas FIR filters use the normal GNU Radio buffer history. I just wanted to point out that blocks can be made to implement reset methods (or accept an inherited version that does nothing), whereas there's no good way to impose reset method on kernels, which are generally just stand alone classes.