gnuradio/gnuradio #1444: Python bindings and Ctrlport tests fail with Thrift 0.10 installed; GR crashes randomly

marcusmueller, 2017-08-27:
After building maint of today with Thrift enabled, compiled by clang:

```
        Start  24: qa_cpp_py_binding
 24/216 Test  #24: qa_cpp_py_binding ....................***Failed    0.59 sec
        Start  25: qa_cpp_py_binding_set
 25/216 Test  #25: qa_cpp_py_binding_set ................***Failed    0.64 sec
        Start  26: qa_ctrlport_probes
 26/216 Test  #26: qa_ctrlport_probes ...................***Failed    1.40 sec
```

with `cmake -DENABLE_CTRLPORT_THRIFT=Off` (only difference), no random crashes. The same tests aren't included in this case, so no comparison can be done here.

Tested on two independent Fedora 26 x86_64 machines. Fedora 26 ships Thrift 0.10.

jmcorgan, 2017-08-27:
So, this code is essentially unmaintained; there doesn't seem to be anybody that wants to continue developing on it. It's likely we'll deprecate it with 3.8, build up an alternative to it during the 3.8->3.9 time period, then finally excise it for 3.9.

marcusmueller, 2017-08-27:
Cool with me, definitely :)

I wonder what makes GR crash with Thrift 0.10, but I haven't been figuring it out, as elbow deep as I'm currently inside GDB. Might simply be that fedora Thrift was built with some non-explicit ABI assumptions that GCC fulfills, but my clang doesn't. Might mean no thrift for me unless I really need it (and then I'll have to try GCC and/or a manually downgraded Thrift).

marcusmueller, 2021-03-21:
should be solved

closed 2021-03-21 by marcusmueller