gnuradio/gnuradio #1446: remove tagged stream blocks

osh, 2017-08-27:


marcusmueller, 2017-08-27:
to extend on this: TSBs are giving us a feature we *need*, that is, bursty operation.

But: that's really a feature that the scheduler should offer as an operational mode for blocks (or rather, subgraphs), and not something we bolt on via tags. I appreciate the existence of TSBs as a solution to the problem of implementing the modern packet blocks in gr-digital! We wouldn't be where we are without them, and they're thus definitely a good thing to have around.

But: there's a lot of bugs and inefficiencies, and quite some grief¹ and undocumented contracts connected to TSBs.

The proper way to solve these problems would be to, as comprehensively as possible, document how the scheduler works with TSBs. Sadly, I can't even offer a comprehensive description of how the scheduler _should_ work, not with nor without TSBs.

The act of documentation has been tried (and failed) several times (at least by me; but I might simply not be clever enough). I'm left under the impression that the best way of documenting things would be to take the basic features of the TPB scheduler, implement them, write down the principles under which they operate, add tags to work like they do today, document, add message passing, document. Then you'd have the state of the art pre-TSB, but hopefully without things like the message passing shutdown problems we lugged around for too long, or the tag sorting problems.

My impression then is that it'd be easier to implement a clean bursty operation mode that simply offers the same functionality as TSBs, but without the tags, and without the problems they bring (those being things like hard-to-define-what's-right packet length changes, inexplicit contracting that blocks used in a TSB path must always fully consume and produce in a single pass, and the fact that we're using ring buffers in a place where they have zero advantage).

Clearly, what I describe above is a rewrite of the scheduler. Frankly, that's been overdue a bit (@bastibl would agree), and it's a good idea to start collecting features (and non-features) that we're going to put in there.

<hr> 
¹ and anger, see @osh's _"TSB and bad ML: a personal rage"_, in: _notes of a doge_, 2015ff

777arc, 2017-08-28:
What about a new type of block that operates on single packets/frames/vectors (whatever you want to call it) at a time, instead of in a streaming-mode.  And then a 2nd type of block that acts as the transition from streaming-mode blocks to vector mode blocks.  Similar to how there are RFNoC blocks, and you use a Copy block to go from RFNoC to normal streaming-mode on the cpu.  That transition block could just be a stream-to-vector type thing, or it could be an energy detector or any block that inherently slices up portions of samples into packets.  

It would be similar to having message passing between blocks, but with defined data types (e.g. vector of complex floats) and better support from the scheduler.  It would also be similar to just using vector items currently, but without the issue of the vectors all having to be the same size, or padded with zeros at the end.  

You wouldn't even have to rewrite the scheduler, you could just write a 2nd scheduler that runs in a separate thread and deals with the packet-type blocks.  For plotting stuff, I would love to be able to just throw the x and y data to a plotter via messages or whatever, without having to worry about the lengths matching across input (example use-case would be testing whether curve-fitting looks good).  I think the new Bokeh plotters could be easily extended to do that.

osh, 2017-08-28:
@777arc 
new type of block that operates on single packets/frames/vectors (whatever you want to call it) at a time
> we have these, they are called message passing blocks
2nd type of block that acts as the transition from streaming-mode blocks to vector mode blocks
> this is what eventstream blocks do now and work fine for most purposes
> our scheduler already deals with these two cases in the same thread, providing thread safety between stream handlers and message handlers.

For plotting stuff, I would love to be able to just throw the x and y data to a plotter via messages or whatever
> this is what gr-pyqwt did and what the updated gr-qtgui message ports do now as well

777arc, 2017-08-28:
Do the updated gr-qtgui message ports allow for different length vectors or data with different x-axis?  So like the equivalent of plt.plot(range(10), range(10), range(20), range(20))?

marcusmueller, 2023-12-21:
Not gonna introduce new mechanisms that replace TSBs altogether in the  3.10/3.11(if that ever happens) timeframe, and solved in 4.0

closed 2023-12-21 by marcusmueller