gnuradio/gnuradio #1449: DVB-T TX - GUARD INTERVAL always in 1/32 - 1/16 1/8 1/4 don't work.

owfrappier, 2017-08-28:
Hi,

in DVB-T transmitter guard interval is always 1/32 (bug), even with other values 1/16, 1/8, 1/4, still stay at 1/32 (0)

see :

gr-dtv/lib/dvbt/dvbt_reference_signals_impl.cc

set_tps_bits(37, 36, config.d_guard_interval);

missing switch :

ETSI 4.6.2.8 :
http://www.etsi.org/deliver/etsi_en/300700_300799/300744/01.06.01_40/en_300744v010601o.pdf

Bits s36, s37 Guard interval values (?/TU)
00 1/32
01 1/16
10 1/8
11 1/4

perhaps shoud be add :

      switch (config.d_code_rate_LP) {
        case GI_1_32:
          set_tps_bits(37, 36, 0);
          break;
        case GI_1_16:
           set_tps_bits(37, 36, 1);
          break;
        case GI_1_8:
           set_tps_bits(37, 36, 2);
          break;
        case GI_1_4:
           set_tps_bits(37, 36, 3);
          break;
        default:
          set_tps_bits(37, 33, 0);
          break;
      }

and in dtv_dvbt_reference_signals.xml

seems incorrect value, should be corrected like this :

  <param>
    <name>Guard Interval</name>
    <key>guard_interval</key>
    <type>enum</type>
    <option>
      <name>1/32</name>
      <key>G1_32</key>
      <opt>val:dtv.GI_1_32</opt>
    </option>
    <option>
      <name>1/16</name>
      <key>G1_16</key>
      <opt>val:dtv.GI_1_16</opt>
    </option>
    <option>
      <name>1/8</name>
      <key>G1_8</key>
      <opt>val:dtv.GI_1_8</opt>
    </option>
    <option>
      <name>1/4</name>
      <key>G1_4</key>
      <opt>val:dtv.GI_1_4</opt>
    </option>
  </param>

jmcorgan, 2017-08-28:
@drmpeg

drmpeg, 2017-08-28:
Works fine here. Note that you have to change the CP Length in the OFDM Cyclic Prefixer block to match the guard interval. For example, in 8k mode and 1/16 guard interval, CP Length = 8192/16.

owfrappier, 2017-08-28:
Hi,
Thanks. Works like a charm. Closed

closed 2017-08-28 by owfrappier