gnuradio/gnuradio #1450: DVB-T wrong CELL-ID : only 8 bits but CELL-ID shound be  16bits within 2 DVB-T frames (2X8 bits)

owfrappier, 2017-08-28:
Hi,

see :

wrong cell-id calculation in DVB-T
gr-dtv/lib/dvbt/dvbt_reference_signals_impl.cc

Final draft ETSI EN 300 744 V1.6.1 (2008-09)

4.6.2.10 Cell identifier
The eight bits s40 to s47 are used to identify the cell from which the signal comes from.

CELL-ID : 16 bits (2 X 8 BITS within 2 frames)

First 8 bits : FRAME 1 and 3
cell_id [4], i.e. b15 - b8, shall be transmitted in super-frames with the frame number 1 and 3. 

+ 8 bits : FRAME 2 and 4
the cell_id, i.e. b7 - b0, shall be transmitted in super-frames with the frame number 2 and 4.

jmcorgan, 2017-08-28:
@drmpeg

drmpeg, 2017-08-28:
It's a bug. I'll provide a pull request soon.

drmpeg, 2019-02-25:
Fixed with commit https://github.com/gnuradio/gnuradio/commit/0d9992d492bf501aa9b31d5872dac801dc362ee8

@noc0lour, you can close this issue.

closed 2019-02-25 by noc0lour