gnuradio/gnuradio #1454: digital.generic_mod broken on MacPorts gnuradio-devel

vamsi765, 2017-08-31:
I get the following error when I use digital.generic_mod

`digital.generic_mod(const, False, samp_symb, False, rolloff, False, False)`

`Value "digital.modulate_vector_bc(preamble_modulator .to_basic_block(), (preamble_hex), ([1]))" cannot be evaluated:`

I'm on latest repository from MacPorts:
gnuradio-devel @20170827_0+ctrlport+debug+docs+grc+jack+log4cpp+logging+performance_counters+portaudio+qtgui+sdl+swig+uhd+wavelet+wxgui+zeromq

`> gnuradio-config-info -v
v3.7-MacPorts-devel-git-34933ac6(20170827)`

The same flow graph works on another system installed using MacPort but at previous version:
gnuradio-devel @20170814_0+ctrlport+docs+grc+jack+log4cpp+logging+performance_counters+portaudio+qtgui+sdl+swig+uhd+wavelet+wxgui+zeromq
> gnuradio-config-info -v
v3.7-MacPorts-devel-git-efcfde0e(20170814)

vamsi765, 2017-09-07:
Hi @michaelld ,

Could you look at this issue when you get a chance?

Thanks!

michaelld, 2017-09-07:
@vamsi765 sure. do you have a GRC flowgraph or Python script that shows this issue? that's easiest for me to test, instead of having to create my own. thx!

vamsi765, 2017-09-14:
I updated the ports again today and I didn't face the issue. Thanks!

closed 2017-09-14 by vamsi765

michaelld, 2017-09-14:
Glad it was that simple! Thanks for reporting back.